cesanta/mongoose-os #306: Auto Scroll checkbox for log windows

chall3ng3r, 2017-07-03:
Just started playing around with this awesome framework. 

There's one small issue, the logs windows sometimes follow the new entries but they just stop randomly, so I have to scroll down to see what's happening. 

I'd be great if Auto Scroll checkbox is added to both log windows, and make them stick to latest entries when checked. When not checked, scroll position should stop at current position.

dimonomid, 2017-08-18:
Thanks for reporting!

Instead of adding a checkbox, we fixed autoscroll logic: it autoscrolls if only the window is already on the bottom. To turn off autoscroll, just scroll up a bit.

Implemented in latest, will be included in the upcoming release 1.15.

closed 2017-08-18 by dimonomid

chall3ng3r, 2017-08-20:
Thanks, will download the latest build to test it out.