cesanta/mongoose-os #307: SONOFF (esp8266) flash size: 1MB - not fit in flash!!

StasiukDanylo, 2017-07-03:
Hello.
Today, after flashing SONOFF (esp8266, 1MB flash size) i received msg: 
**"Error: Image 673824 @ 0x100000 will not fit in flash (size 1048576)"**

Previously, the result of this module was successful!
Have you added limitation to the module's memory size?

StasiukDanylo, 2017-07-08:
Does not it matter to anyone?

rojer, 2017-07-12:
we moved things around and the default flash layout is now 4MB.
1MB is still supported, but you need to specify
```
build_vars:
  FLASH_SIZE: 1048576
```

closed 2017-07-12 by rojer

rojer, 2017-07-12:
over to @cpq for UI part

valtersitz, 2017-07-19:
Hi guys,

Reopening this because this is flash size - will not fit related.
I am working with ESP WROOM-02. Followed the previous fix with 

> build_vars:
  FLASH_SIZE: 2097152

as this is a 2MB flash.
But while flashing I get 
`
Running flasher @ 460800...
  Flasher is running
Flash size: 2097152, params: 0x0230 (dio,16m,40m)    // see here the 16m parameter
Deduping...
   619088 @ 0x8000 -> 368640
   131072 @ 0xdb000 -> 0
     4096 @ 0x1fb000 -> 0
Writing...
     4096 @ 0x0
     4096 @ 0x7000
   368640 @ 0x8000
     4096 @ 0x1fc000
Wrote 380928 bytes in 8.38 seconds (355.19 KBit/sec)
Verifying...
     2656 @ 0x0
     4096 @ 0x7000
   619088 @ 0x8000
   131072 @ 0xdb000
     4096 @ 0x1fb000
      128 @ 0x1fc000
Booting firmware...
All done!
`

and after booting: 
`[Jul 19 20:26:42.808] rBoot v1.2.1 - richardaburton@gmail.com
[Jul 19 20:26:42.809] Flash Size:   16 Mbit
[Jul 19 20:26:42.809] Flash Mode:   DIO
[Jul 19 20:26:42.809] Flash Speed:  40 MHz
[Jul 19 20:26:42.809] rBoot Option: Big flash
[Jul 19 20:26:42.810] 
[Jul 19 20:26:42.810] Writing default boot config @ 0x7000.
[Jul 19 20:26:42.840] Rom 0 is bad.
[Jul 19 20:26:42.924] Booting rom 1 (0x200000).
[Jul 19 20:26:42.924] 
[Jul 19 20:26:42.924] rBoot v1.2.1 - richardaburton@gmail.com
[Jul 19 20:26:42.924] Flash Size:   16 Mbit
[Jul 19 20:26:42.924] Flash Mode:   DIO
[Jul 19 20:26:42.925] Flash Speed:  40 MHz
[Jul 19 20:26:42.925] rBoot Option: Big flash
[Jul 19 20:26:42.925] 
[Jul 19 20:26:42.988] Booting rom 1 (0x200000).
[Jul 19 20:26:42.988] 
[Jul 19 20:26:42.988] rBoot v1.2.1 - richardaburton@gmail.com
[Jul 19 20:26:43.009] Flash Size:   16 Mbit
[Jul 19 20:26:43.009] Flash Mode:   DIO
[Jul 19 20:26:43.009] Flash Speed:  40 MHz
[Jul 19 20:26:43.009] rBoot Option: Big flash
[Jul 19 20:26:43.009] 
[Jul 19 20:26:43.073] Booting rom 1 (0x200000).
[Jul 19 20:26:43.073] 
`
to the infinite and beyond... 

I went looking into the bootloader, rBoot, found the Big Flash part but I could not find any way to modify the 

>  Rom 0 is bad

Would you have any fix for this? Did I miss something?