cesanta/mongoose-os #309: Building new firmware problem

Sparow199, 2017-07-07:
Hello ,

I have the last version of `mos`,  month ago, i had made a project called [MQ-135](https://github.com/Sparow199/MQ135_MONGOOSE_OS) and i had test it and it worked.

Today, i cloned ma repository, and i tried it , it doesn't work !!! 

I tried with Mac OS and Ubuntu but no way.

I flashed my esp8266 with default firmware then i uploaded my `init.js` and `api_mq.js` files, it works !!!

riless, 2017-07-07:
Just read the error.

`mjs_exec_file` does not take `generate_jsc` as a third parameter [anymore](https://github.com/cesanta/mjs/commit/88b387b60595f57db0fb6b6b94b682dac666865d#diff-dfd500dfec3905a96aff2852116f6fe1L8084).

You should replace [this line ](https://github.com/Sparow199/MQ135_MONGOOSE_OS/blob/master/src/main.c#L40)by:

```
mjs_err_t err = mjs_exec_file(mjs, "init.js", NULL);
mjs_set_generate_jsc(mjs, 1);
```

Sparow199, 2017-07-07:
@riless thank you for the correction, i modified the issue in order to simply it and focus on the main problem, because it doesn't work  with whole projects not only mines.

cpq, 2017-07-15:
We have refactored mos architecture a bit, now it has apps and libraries.

@Sparow199 do you mind contributing that example under the Apache 2.0 license, per https://mongoose-os.com/docs/overview/apps.html#contributing-an-app-or-library ?

If you're fine with that, I'll move your app and fix the build.

Sparow199, 2017-07-15:
@cpq thank you for reply, i'll try the build process next Monday  and i'll keep you informed about issue.
 I'll contribute with my library soon.

Sparow199, 2017-07-17:
@riless @cpq , it works with last MOS update.
THANK YOU

cpq, 2017-07-18:
@Sparow199 do you mind me adding your app to the mongoose-os-apps collection under the Apache 2.0 license?

Sparow199, 2017-07-18:
@cpq  it will be available soon

cpq, 2017-07-21:
https://github.com/mongoose-os-libs/mq135 - you have admin rights.
Feel free to import there.
Don't forget to set the description, yourself as an author, and Apache2.0 license, please.

Thank you!

closed 2017-07-21 by cpq