cesanta/mongoose-os #312: MQTT SSL issue

helscream, 2017-07-12:
I am running Mosquitto on Raspberry PI and it is configured for SSL connection on port 1883 (I tired port 8883 and it also not working)
here is the mosquitto.conf

>pid_file /var/run/mosquitto.pid
>persistence true
>persistence_location /var/lib/mosquitto/
>log_dest file /var/log/mosquitto/mosquitto.log
>include_dir /etc/mosquitto/conf.d
>log_type error
>log_type warning
>log_type notice
>log_type information
>connection_messages true
>log_timestamp true

>port 1883
>cafile /etc/mosquitto/ca_certificates/ca.crt
>certfile /etc/mosquitto/certs/sens.crt
>keyfile /etc/mosquitto/certs/sens.key
>require_certificate true

>allow_anonymous false
>password_file /etc/mosquitto/passwd_mqtt

In Mosquitto logs i see the failed connection like this 
>1499819744: Socket error on client <unknown>, disconnecting.
>1499819805: New connection from 192.168.10.1 on port 1883.
>1499819805: Socket error on client <unknown>, disconnecting.
>1499819869: New connection from 192.168.10.1 on port 1883.

before I go to MongooseOS configuration, I am using mqtt-spy for testing my service running on Raspberry PI and It connects successfully with all SSL and user authentication
However, with MongooseOS i am using the exact same CA.crt, client.crt and client.key but it gives me this error from MongooseOS log
>mg_ssl_if_mbed_err   0x3ffb3d80 SSL error: -9984
>mgos_mqtt_ev         MQTT Connect (0)
>mgos_mqtt_ev         MQTT Disconnect
>mqtt_global_reconnec MQTT connecting after 62266 ms

Mongoose configuration file 
"mqtt": {
    "enable": true,
    "server": "sens.live:1883",
    "client_id": "esp32",
    "user": "esp32",
    "pass": "123456",
    "reconnect_timeout_min": 2,
    "reconnect_timeout_max": 60,
    "ssl_cert": "esp32.crt",
    "ssl_key": "esp32.key",
    "ssl_ca_cert": "ca.crt",
    "ssl_cipher_suites": "",
    "ssl_psk_identity": "",
    "ssl_psk_key": "",
    "clean_session": true,
    "keep_alive": 60,
    "will_topic": "",
    "will_message": ""
  },
What I am doing wrong?

cpq, 2017-07-12:
Error code 9984 is MBEDTLS_ERR_X509_CERT_VERIFY_FAILED

helscream, 2017-07-12:
Thanks for clearly stating the message.
I am a bit confused. I am using same certificates used in mqtt-spy. may be I am not uploading them to FS properly or some thing
I used Windows command line to download the certificates using 
>mos put ca.crt
>mos put esp32.crt
>mos put esp32.key

helscream, 2017-07-13:
Yeah, some thing is wrong with my certificates 😄

closed 2017-07-13 by helscream