cesanta/mongoose-os #313: Using both AWS and Dallas OneWire temp sensor causes core dump

zendes, 2017-07-12:
I have a nodemcu esp-12e that is connected to a dallas ds18b20 onewire temperature sensor
reading the temperature from the sensor works as expected

commenting out that code, connecting the module to aws iot and publishing a message works as expected

when uncommenting the onewire libraries in init.js (api_arduino_{onewire,dallas_temp}.js) the module crashes with an exception (logs below)

is this a known issue?

[Jul 11 21:50:39.798] rBoot v1.2.1 - richardaburton@gmail.com                                                                                                                         
[Jul 11 21:50:39.798] Flash Size:   32 Mbit                                                                                                                                           
[Jul 11 21:50:39.798] Flash Mode:   DIO                                                                                                                                               
[Jul 11 21:50:39.798] Flash Speed:  40 MHz                                                                                                                                            
[Jul 11 21:50:39.799] rBoot Option: Big flash                                                                                                                                         
[Jul 11 21:50:39.799]                                                                                                                                                                 
[Jul 11 21:50:39.803] Writing default boot config @ 0x7000.                                                                                                                           
[Jul 11 21:50:39.899] Booting rom 0 (0x100000).                                                                                                                                       
[Jul 11 21:50:40.113]       n  r  n|  l lll`  b     | r  l n   n   l`   r l l     l`   r l l     l`   r l      ll`     n   n lrlmode : sta(18:fe:34:a5:4e:e3)                         
[Jul 11 21:50:40.151] add if0                                                                                                                                                         
[Jul 11 21:50:40.152]                                                                                                                                                                 
[Jul 11 21:50:40.158] esp_mgos_init2       temperature 1.0.0 (20170712-045032/???)                                                                                                    
[Jul 11 21:50:40.165] esp_mgos_init2       Mongoose OS Firmware 2017071204 (20170712-045032/???)                                                                                      
[Jul 11 21:50:40.175] esp_mgos_init2       SDK 2.1.0(ce90efd), RAM: 52472 total, 49596 free                                                                                           
[Jul 11 21:50:40.175] esp_print_reset_info Reset cause: 6 (sys reset)                                                                                                                 
[Jul 11 21:50:40.180] mgos_vfs_dev_open    sysflash () -> 0x3ffefe5c                                                                                                                  
[Jul 11 21:50:40.190] mgos_vfs_mount       Mount SPIFFS @ / (dev 0x3ffefe5c, opts {"addr": 32768, "size":262144}) -> 0x3ffefe6c                                                       
[Jul 11 21:50:40.252] mgos_vfs_mount       /: size 233681, used: 94376, free: 139305                                                                                                  
[Jul 11 21:50:40.343] mgos_sys_config_init MAC: 1AFE34A54EE3                                                                                                                          
[Jul 11 21:50:40.347] mgos_sys_config_init WDT: 30 seconds                                                                                                                            
[Jul 11 21:50:40.348] scandone                                                                                                                                                        
[Jul 11 21:50:40.355] del if0                                                                                                                                                         
[Jul 11 21:50:40.355] usl                                                                                                                                                             
[Jul 11 21:50:40.355] mode : null                                                                                                                                                     
[Jul 11 21:50:40.355] mgos_wifi_set_mode   WiFi mode: STA                                                                                                                             
[Jul 11 21:50:40.367] mode : sta(18:fe:34:a5:4e:e3)                                                                                                                                   
[Jul 11 21:50:40.367] add if0                                                                                                                                                         
[Jul 11 21:50:40.367] WPA2 ENTERPRISE VERSION: [v2.0] disable                                                                                                                         
[Jul 11 21:50:40.367] mgos_wifi_setup_sta  WiFi STA: Connecting to Old New York                                                                                                       
[Jul 11 21:50:40.375] mgos_http_server_ini HTTP server started on [80]                                                                                                                
[Jul 11 21:50:40.381] mg_rpc_channel_uart  0x3fff1184 UART0                                                                                                                           
[Jul 11 21:50:40.388] mgos_init            Init done, RAM: 52472 total, 44548 free, 44288 min free                                                                                    
[Jul 11 21:50:41.388] Device ID:  esp8266_temp_A54EE3                                                                                                                                 
[Jul 11 21:50:41.412] mongoose_poll        New heap free LWM: 960                                                                                                                     
[Jul 11 21:50:42.414] online: false mqtt online: false                                                                                                                                                                                                                                                                                                                                                                                 
[Jul 11 21:50:43.413] online: false mqtt online: false
[Jul 11 21:50:43.421] scandone
[Jul 11 21:50:43.424] state: 0 -> 2 (b0)
[Jul 11 21:50:43.430] state: 2 -> 3 (0)
[Jul 11 21:50:43.435] state: 3 -> 5 (10)
[Jul 11 21:50:43.435] add 0
[Jul 11 21:50:43.435] aid 4
[Jul 11 21:50:43.435] cnt
[Jul 11 21:50:43.462]
[Jul 11 21:50:43.462] connected with Old New York, channel 11
[Jul 11 21:50:43.462] dhcp client start...
[Jul 11 21:50:43.462] mgos_wifi_on_change_ Wifi: connected
[Jul 11 21:50:44.057] ip:10.0.0.119,mask:255.255.255.0,gw:10.0.0.1
[Jul 11 21:50:44.057] mgos_wifi_on_change_ WiFi: ready, IP 10.0.0.119, DNS 10.0.0.1
[Jul 11 21:50:44.073] mqtt_global_reconnec MQTT connecting after 2169 ms
[Jul 11 21:50:44.414] online: true mqtt online: false
[Jul 11 21:50:45.069] mgos_sntp_query      SNTP query to pool.ntp.org
[Jul 11 21:50:45.207] mgos_sntp_ev         SNTP reply from 216.229.0.50: time 1499835045.544627, local 5.382091, delta 1499835040.162536
[Jul 11 21:50:45.426] online: true mqtt online: false
[Jul 11 21:50:46.264] mqtt_global_connect  MQTT connecting to a2ysalqhatzmmu.iot.us-west-2.amazonaws.com:8883
[Jul 11 21:50:46.393] E:M 1024 (129 blocks)
[Jul 11 21:50:46.485] E:M 816 (103 blocks)
[Jul 11 21:50:46.489] E:M 256 (33 blocks)
[Jul 11 21:50:46.512]
[Jul 11 21:50:46.512] Exception 29 @ 0x4000e1b2, vaddr 0x00000000
[Jul 11 21:50:46.512]  A0: 0x4029050f  A1: 0x3ffff700  A2: 0x00000000  A3: 0x00000000
[Jul 11 21:50:46.513]  A4: 0x00000100  A5: 0x00000000  A6: 0x00000100  A7: 0x00000010
[Jul 11 21:50:46.513]  A8: 0x3ffe83cc  A9: 0x00000190 A10: 0x00000000 A11: 0x636f6c62
[Jul 11 21:50:46.513] A12: 0x00000000 A13: 0x00000100 A14: 0x00000000 A15: 0x3fff903f
[Jul 11 21:50:46.534]
[Jul 11 21:50:46.534] (exc SP: 0x3ffff560)

cpq, 2017-07-12:
Looks like OOM.
Could you log free RAM as well please?

zendes, 2017-07-13:
I've gotten another interesting error message:
heap integrity broken: block links don't match: 3473 -> 3604, but 3604 -> 3328

i've attached two logs with free ram (timer at 100ms, just doing print Sys.free_ram())

[err1.txt](https://github.com/cesanta/mongoose-os/files/1143669/err1.txt)
[err2.txt](https://github.com/cesanta/mongoose-os/files/1143668/err2.txt)

zendes, 2017-07-13:
looks like it's indeed free RAM. if i comment out all load statements besides sys, timer, onewire and dallas_temp it does succesfully connect.  

looks like the initial mqtt connect takes about 11k of ram. do you have any tips on how to load libraries after that? if i add load statements later in the code, they don't seem to work (i get an error about not being able to read the file)

[Jul 12 17:10:33.307] Free RAM: 26676
[Jul 12 17:10:33.380] mqtt_global_connect  MQTT connecting to a2ysalqhatzmmu.iot.us-west-2.amazonaws.com:8883
[Jul 12 17:10:33.416] Free RAM: 20332
[Jul 12 17:10:40.288] mongoose_poll        New heap free LWM: 1552
[Jul 12 17:10:40.325] Free RAM: 10192
[Jul 12 17:10:40.729] pm open,type:2 0
[Jul 12 17:10:40.826] Free RAM: 9688
[Jul 12 17:10:44.744] mgos_mqtt_ev         MQTT Connect (1)
[Jul 12 17:10:44.808] mgos_mqtt_ev         MQTT CONNACK 0
[Jul 12 17:10:44.826] Free RAM: 23560

this is init.js:
load('api_sys.js');
load('api_timer.js');
load('api_arduino_onewire.js');
load('api_arduino_dallas_temp.js');

Timer.set(100, true, function() {
  print('Free RAM:', Sys.free_ram());
}, null);

and mos.yml bits
filesystem:
  - fs

config_schema:
  - ["debug.level", 2]
  - ["device.id", "esp8266_temp_??????"]
  - ["mqtt.enable", true]
  - ["mqtt.server", "a2ysalqhatzmmu.iot.us-west-2.amazonaws.com:8883"]
  - ["mqtt.ssl_ca_cert", "ca-verisign-g5.crt.pem"]
  - ["mqtt.ssl_cert", "aws-iot-85cd6d83d5.crt.pem"]
  - ["mqtt.ssl_key", "aws-iot-85cd6d83d5.key.pem"]
  - ["mqtt.ssl_cipher_suites", "TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256"]
  - ["wifi.ap.enable", false]
  - ["wifi.sta.enable", true]
  - ["wifi.sta.pass", "xxxxxxxx"]
  - ["wifi.sta.ssid", "Old New York"]

libs:
  - origin: https://github.com/mongoose-os-libs/mjs
  - origin: https://github.com/mongoose-os-libs/mqtt
  - origin: https://github.com/mongoose-os-libs/arduino-onewire
  - origin: https://github.com/mongoose-os-libs/arduino-dallas-temperature

cpq, 2017-07-13:
We need to restore mmap and JS pre-compile.
@dimonomid please handle!

dimonomid, 2017-07-18:
@zendes , we've restored mmap on esp platform, this frees 5+ K of RAM. Please try now with the latest firmware (if you use release, you'll need to update to latest with `mos update latest`)

zendes, 2017-07-19:
Will do. I'll get back to you in about a day with the results.

zendes, 2017-07-19:
looks like that fixed it. i can now load all libs at the top, and publish a ds18b20 reading to aws iot without the device rebooting or crashing. thnx!

closed 2017-07-19 by zendes