cesanta/mongoose-os #314: Can't flash ESP32

ca0abinary, 2017-07-13:
Ran command (also without changing the baud rate, but 230400 works for the ESP32 Download Tool, also tried every baud rate from 115200 up):
`.\mos flash --esp-baud-rate 230400 -v 4 --logtostderr`

Got results:
```
Loaded default/esp32 version 1.0 (20170713-200658/???)
Using port COM4
I0713 18:08:59.241452   20820 ourutil.go:15] Using port COM4
I0713 18:08:59.241452   20820 flash_params.go:104]  -> -1, -1, -1
Opening COM4...
I0713 18:08:59.241452   20820 report.go:13] Opening COM4...
Connecting to ESP32 ROM, attempt 1 of 10...
I0713 18:08:59.309996   20820 report.go:13] Connecting to ESP32 ROM, attempt 1 of 10...
I0713 18:09:00.913205   20820 rom_client.go:170] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0713 18:09:00.914208   20820 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0713 18:09:01.116285   20820 rom_client.go:170] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0713 18:09:01.116285   20820 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0713 18:09:01.318396   20820 rom_client.go:153] Sync #1 failed: error reading sync response #1: error reading response: error reading: EOF
Connecting to ESP32 ROM, attempt 2 of 10...
I0713 18:09:01.318396   20820 report.go:13] Connecting to ESP32 ROM, attempt 2 of 10...
I0713 18:09:02.922782   20820 rom_client.go:170] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0713 18:09:02.923268   20820 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0713 18:09:03.123884   20820 rom_client.go:170] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0713 18:09:03.124385   20820 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0713 18:09:03.130384   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.130882   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.131383   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.131883   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.133387   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.133883   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.134386   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.134886   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.135387   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.135387   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.136386   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.136888   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.138384   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.138883   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.139384   20820 slip.go:46] <= (12) 010804000720555500000000
I0713 18:09:03.139384   20820 rom_client.go:214] <= {cmd:Sync(8) value:1431642119 ok:true lastError:0 body(4):"00000000"}
  Connected
I0713 18:09:03.139885   20820 report.go:13]   Connected
Running flasher @ 230400...
I0713 18:09:03.140384   20820 report.go:13] Running flasher @ 230400...
I0713 18:09:03.140884   20820 rom_client.go:379] RunStub: params: 1 @ 0x40090000; code: 4244 @ 0x40090004, entry @ 400907a8; data: 36 @ 0x3ffd8018
I0713 18:09:03.142383   20820 rom_client.go:305] memWrite(0x40090000, 4, 0x00000000)
I0713 18:09:03.142885   20820 rom_client.go:170] => {cmd:MemWriteStart(5) csum:0 arg(16):"04000000010000000010000000000940"}
I0713 18:09:03.143387   20820 slip.go:90] => (24) 000510000000000004000000010000000010000000000940
I0713 18:09:03.147383   20820 slip.go:46] <= (12) 010504000720555500000000
I0713 18:09:03.147883   20820 rom_client.go:214] <= {cmd:MemWriteStart(5) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.148383   20820 rom_client.go:170] => {cmd:MemWriteBlock(7) csum:104 arg(20):"0400000000000000000000000000000000840300"}
I0713 18:09:03.148883   20820 slip.go:90] => (28) 00071400680000000400000000000000000000000000000000840300
I0713 18:09:03.152886   20820 slip.go:46] <= (12) 010704000720555500000000
I0713 18:09:03.153896   20820 rom_client.go:214] <= {cmd:MemWriteBlock(7) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.153896   20820 rom_client.go:170] => {cmd:MemWriteFinish(6) csum:0 arg(8):"0100000000000000"}
I0713 18:09:03.154898   20820 slip.go:90] => (16) 00060800000000000100000000000000
I0713 18:09:03.157897   20820 slip.go:46] <= (12) 010604000720555500000000
I0713 18:09:03.158895   20820 rom_client.go:214] <= {cmd:MemWriteFinish(6) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.158895   20820 rom_client.go:305] memWrite(0x3ffd8018, 36, 0x00000000)
I0713 18:09:03.158895   20820 rom_client.go:170] => {cmd:MemWriteStart(5) csum:0 arg(16):"2400000001000000001000001880fd3f"}
I0713 18:09:03.159898   20820 slip.go:90] => (24) 00051000000000002400000001000000001000001880fd3f
I0713 18:09:03.163897   20820 slip.go:46] <= (12) 010504000720555500000000
I0713 18:09:03.164901   20820 rom_client.go:214] <= {cmd:MemWriteStart(5) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.164901   20820 rom_client.go:170] => {cmd:MemWriteBlock(7) csum:82 arg(52):"24000000000000000000000000000000ae060940c7060940e406094002070940..."}
I0713 18:09:03.164901   20820 slip.go:90] => (60) 000734005200000024000000000000000000000000000000ae060940c7060940...
I0713 18:09:03.171896   20820 slip.go:46] <= (12) 010704000720555500000000
I0713 18:09:03.172905   20820 rom_client.go:214] <= {cmd:MemWriteBlock(7) value:1431642119 ok:true lastError:0 body(4):"00000000"}
I0713 18:09:03.172905   20820 rom_client.go:170] => {cmd:MemWriteFinish(6) csum:0 arg(8):"0100000000000000"}
I0713 18:09:03.172905   20820 slip.go:90] => (16) 00060800000000000100000000000000
I0713 18:09:03.176901   20820 slip.go:46] <= (12) 010604000720555500000000
[... 42 lines cut ...]
I0713 18:09:03.761044   20820 flasher_client.go:102] Echo(8) [3063604278]
I0713 18:09:03.761044   20820 slip.go:90] => (1) 08
I0713 18:09:03.762044   20820 slip.go:90] => (4) 36e49ab6
I0713 18:09:03.762044   20820 flasher_client.go:217] <= c0
I0713 18:09:03.763056   20820 flasher_client.go:217] <= 36
I0713 18:09:03.763056   20820 flasher_client.go:217] <= e49ab6c0c000c0
I0713 18:09:03.814045   20820 slip.go:90] => (0)
I0713 18:09:03.814045   20820 flasher_client.go:102] Echo(8) [3829376135]
I0713 18:09:03.815045   20820 slip.go:90] => (1) 08
I0713 18:09:03.815045   20820 slip.go:90] => (4) 87a43fe4
I0713 18:09:03.816045   20820 flasher_client.go:217] <= c0
I0713 18:09:03.816045   20820 flasher_client.go:217] <= 87
I0713 18:09:03.816045   20820 flasher_client.go:217] <= a43fe4c0c000
I0713 18:09:03.817046   20820 flasher_client.go:217] <= c0
I0713 18:09:03.868075   20820 main.go:212] Error: failed to run flasher: failed to talk to flasher: flasher did not respond
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:226: flasher did not respond
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:94: failed to talk to flasher
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:66:
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/common.go:65: failed to run flasher
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flash.go:43:
/Users/mos/go/src/cesanta.com/mos/flash.go:128:
/Users/mos/go/src/cesanta.com/mos/main.go:125:
Error: failed to run flasher: failed to talk to flasher: flasher did not respond```

I'm all out of ideas, has anyone else had this experience?

nccchirag, 2017-07-14:
Same issue

**Environment**
OS - Win x64 
Dev board - Geekworm | Easy Kit ESP32-B1 | ESP-WROOM-32 (4M)
USB UART - WCH CH340C

**Command**
mos flash esp32 --port COM7 --esp-baud-rate 115200 --esp-flash-params=dio,32m,40m -v 4 --logtostderr

**Error**
Error: failed to run flasher: failed to talk to flasher: flasher did not respond

Can we use esptool to flash mongoose ?

**Note:** flashing works fine with Arduino 1.8.3

ca0abinary, 2017-07-14:
As an update, I think Windows (or the serial driver used by windows) might be the factor here. When I connected this to my mac it was up and running in less than a minute. The serial bridge on my device claims to be a Silicon Labs CP210x but the chip itself looks to have no laser engraving, so it might be a fake.

nccchirag, 2017-07-15:
I tried both on windows and linux (Ubuntu 16.04 x64), no luck still. The driver seems to be working fine as I can flash via Arduino and also through esptool (on windows) 

I wonder if there is a manual way of flashing mongoose OS via esptool ( @cpq kindly guide )

**Note:** Driver works on linux too ( basic Tx/Rx works on minicom )

nccchirag, 2017-07-15:
**solved it** - used another [USB - UART](http://www.hardkernel.com/main/products/prdt_info.php?g_code=G134111883934)

Problem is with the USB-UART IC CH340C in the Dev board - Geekworm | Easy Kit ESP32-B1

@ca0abinary - you may close the issue

cpq, 2017-07-15:
Thanks for the background!

closed 2017-07-15 by cpq

manikawnth-abg, 2017-08-06:
Just an update for the windows users

I installed the drivers as mentioned USB-to-UART drivers from below site and everything worked as expected

1. Install the drivers
2. Reboot the PC
3. Flash with esp32 and set the wifi

https://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx