cesanta/mongoose-os #320: Enhancement: Modify Power Transmission Level for ESP devices

applecrusher, 2017-07-21:
Is this something that can be added or is something that can be done already. 

For example, I like to take advantage of these 3 transmission levels for the ESP8266:

Tx802.11b, CCK 11Mbps, P OUT=+17dBm - 170 - mA
Tx 802.11g, OFDM 54Mbps, P OUT =+15dBm - 140 - mA
Tx 802.11n, MCS7, P OUT =+13dBm - 120 - mA

Thanks! Let me know what is required to get this done, if it's not too difficult I can help out with this.

cpq, 2017-07-22:
If that is driven by the `system_phy_set_max_tpw ` SDK API, then you can use it directly from your C sources.

If you're using JS, FFI this function and rebuild the FW.
Feel free to send a PR that adds FFI to the https://github.com/mongoose-os-libs/mjs/blob/master/esp8266/fs/api_esp8266.js

applecrusher, 2017-07-23:
Thank you!

So are you saying it is as easy adding the following below:

```
let ESP8266 = {
  ...
  // ## **`setOutputPower(uint8_t val)`**
  // Power output is a number between 0 and 82
  // Return value:  does not return.
  setOutputPower: ffi(' void system_phy_set_max_tpw(uint8_t)')
};

```
Or am I missing something more, like having to import some c code?

cpq, 2017-07-24:
That should be it.
Did you try it, how does that work for you?

applecrusher, 2017-07-24:
Not yet. But I will have some time tonight and will let you know.

applecrusher, 2017-07-25:
I put it in my project init.js directly, not in the ESP8266 js file. Please let me know if that makes a difference.

I wrote the following:

```
let setOutputPower = ffi('void system_phy_set_max_tpw(int)');
setOutputPower(20);
```

The error I got:

`MJS error: bad ffi signature: "void system_phy_set_max_tpw(int)": dlsym('system_phy_set_max_tpw') failed
`

closed 2017-09-14 by applecrusher