cesanta/mongoose-os #324: When I try to install mos-esp8266, I received: "failed to fetch: 404 Not Found"

aderbas, 2017-07-27:
I followed the tutorial, in section "Installation on Linux or MacOS" but when I try command to flash on esp8266 the link to download firmware is broken.

```
$ /root/.mos/bin/mos flash mos-esp8266
Fetching https://mongoose-os.com/downloads/mos-esp8266-1.11.zip...
Error: https://mongoose-os.com/downloads/mos-esp8266-1.11.zip: failed to fetch: 404 Not Found
```

chamalC, 2017-07-27:
I have the same issue 

Fetching https://mongoose-os.com/downloads/mos-esp8266-1.10.zip...
Error: https://mongoose-os.com/downloads/mos-esp8266-1.10.zip: failed to fetch: 404 Not Found

aderbas, 2017-07-27:
The file name is wrong, _mos-esp8266-1.10.zip_ the correct is **_esp8266-1.10.zip_**
see: https://mongoose-os.com/downloads/esp8266-1.10.zip
But that does not solve our problem.

chamalC, 2017-07-27:
then I tried the GUI with mac OS and it also returns an error in the log

E0727 20:14:11.128120   11104 ui.go:467] Calling: FS.ListExt
Server Error: {"error":"Device is not connected"}
Loaded https://mongoose-os.com/downloads/api.json
E0727 20:14:20.715404   11104 reconnect_wrapper.go:93] [reconnectWrapperCodec to /dev/cu.HBS-730-SerialPort; connect in 2.00s] connection error: open /dev/cu.HBS-730-SerialPort: resource busy
open /dev/cu.HBS-730-SerialPort: resource busy
/Users/mos/go/src/cesanta.com/common/go/mgrpc/codec/serial.go:76: 
/Users/mos/go/src/cesanta.com/common/go/mgrpc/mgrpc.go:154: 
/Users/mos/go/src/cesanta.com/common/go/mgrpc/mgrpc.go:201:

cpq, 2017-08-16:
please update, `mos update`
Then, make sure there is no other mos process is running.
Reopen if still an issue.

closed 2017-08-16 by cpq