cesanta/mongoose-os #328: UART.read not working properly on ESP32 (data "shifted" by one)

daniele-salvagni, 2017-08-04:
### My setup

I have an `ESP-WROOM-32` module and I need to communicate with another device via UART. Before moving on to the the actual device I'm testing my code wiring my ESP32 to a **USB-Serial** converter with a `CH340G` chip. Only `TX`, `RX` and `GND` are connected.

This is my configuration:

```
let uartNo = 2;
UART.setConfig(uartNo, {
  baudRate: 1200,
  numDataBits: 8,
  numStopBits: 2,
  parity: 0
});
```

### What does and what doesn't work

- Sending data `ESP32 -> PC (UART.write)` is fine
- Receiving data `PC -> ESP32 (UART.read)` is **not**

Code used to receive data:

```
UART.setDispatcher(uartNo, function(uartNo, ud) {
  let ra = UART.readAvail(uartNo);
  if (ra > 0) {
    let data = UART.read(uartNo);
    print("Received UART data:", data);
  }
}, null);
```

I'm using HTerm to send data and here is what happens (numbers are in hexadecimal):

    Sent from PC:         01 02 03 04 05 06 07 08 09 0A 32 33
    Received on ESP32:    80 80 81 81 82 82 83 83 84 84 99 99

Now, to illustrate better what's happening let's take `0x32` and `0x33` which are both received as `0x99` and look at the binary representation:

    HEX    BINARY         SHIFTED
    0x32   0011 0010  >>  -001 1001
    0x33   0011 0011  >>  -001 1001
    0x99   1001 1001      1001 1001

If we *shift the two numbers to the right by 1* we get the same number which could give an insight of what's happening.

### Logic levels

I did try to check with a logic analyzer and everything looks fine, so the problem should be on the software side:

- `PC -> ESP32`

  [image]

- `ESP32 -> PC`

  [image]

Everything looks fine here.

I found this user with the same issue so it should be reproducible: https://forum.mongoose-os.com/discussion/1367/uart-read-strange-output

rojer, 2017-08-05:
thanks for the detailed report. turns out, ESP32 has a hardware bug in UART receiver which prevents 2 bit stop mode from working properly. when testing, i only tested _transmitting_ in 7O2 mode, not receiving... the fix has been sent and will land in the public repo soon.

closed 2017-08-05 by rojer

daniele-salvagni, 2017-08-06:
I can confirm it is now fixed, thanks!

rajibgupta, 2018-08-08:
How did you make it work ? @daniele-salvagni

DevendraDevadiga, 2021-10-08:
Hi, I am also facing same issue. Is this working for anyone? i.e sending data from PC to ESP32 not working..

scaprile, 2021-10-08:
> 
> 
> Hi, I am also facing same issue. Is this working for anyone? i.e sending data from PC to ESP32 not working..

I've been using the UART for years, many developers have. This issue has been closed for over 3 years now.
I you have a problem using it I suggest you get proper tools and gather proper debugging information, then ask for advice at [the forum](https://community.mongoose-os.com/).