cesanta/mongoose-os #336: FR: dns server

jocodeinc, 2017-08-25:
When connected to device (e.g., ESP8266_12345), user can access index.html (similar to wifi-setup app) using fake domain name (e.g., setup.mongoose.com) rather than 192.168.4.1

It'll make it easier for user (non-techie) to interact with the setup rather than using IP address.

cpq, 2017-09-13:
This is important for usability.
@ruslanvaliullin take on this one.

wormyrocks, 2017-09-17:
This and softAP redirect are also crucial to my project. As far as I can tell neither of the existing captive portal examples for the esp32 will even redirect websites that use https.

closed 2017-12-15 by rojer

dimonomid, 2017-12-15:
Resolved with https://github.com/mongoose-os-libs/wifi/commit/ad19f7ac63b484ccd3047b7c9823589dbef021b1