cesanta/mongoose-os #338: FR: Add Cfg.set(path) in api_config.js

jocodeinc, 2017-08-27:
Currently, there's a way to retrieve config value using [Cfg.get(path)](https://mongoose-os.com/docs/libraries/core/config.html#Cfg.get), but not a way to edit an existing configuration.

**Use case**: update `wifi.sta.enable` to _false_ and `wifi.ap.enable` to _true_. At present, mJS code would involve 
```
let obj = JSON.parse(File.read('conf9.json')); // Read conf9.json file into JS object
obj.wifi.sta.enable = false;
obj.wifi.ap.enable = true;
File.write(JSON.stringify(obj), 'conf9.json'); // Write updated object into conf9.json
Sys.reboot(0); // Reboot system to save settings
```
Suggested program could be documented as the following:
**Cfg.set(config, variable)**
Set the `config value` in conf9.json by the `configuration variable`, where only simple types are currently allowed: strings, ints, booleans, doubles. Objects are not yet supported.
```
Example:
  load('api_config.js');
  Cfg.set('wifi.sta.enable', false); // Disable STA mode
  Cfg.set('wifi.ap.enable', true); // Enable AP mode
  Sys.reboot(0); // Reboot system to save settings
```
Note: It would be great to be able to update any existing configuration but at minimum, this functionality should cover configurations listed in conf9.json

cpq, 2017-09-13:
@ruslanvaliullin that FR makes perfect sense to me.
Please implement.

wormyrocks, 2017-09-17:
hi,

I implemented this for my own use. anyone who wants to use my code is welcome:

C code:
```
void save_config(const struct mgos_conf_entry *e, void *newval){
  struct sys_config *cfg = get_cfg();
  int pass = 0;
  char *vp = (((char *) cfg) + e->offset);
  if (e->type == CONF_TYPE_STRING){
    char **sp = (char **) vp;
    if (*sp != NULL) free(*sp);
    *sp = (char *) malloc(strlen(newval)+1);
    strcpy(*sp, newval);
    pass = 1;
  }else if (e->type == CONF_TYPE_INT || e->type == CONF_TYPE_BOOL){
    int *ip = (int *) vp;
    *ip = strtol(newval, NULL, 10);
    pass = 1;
  }
  if (pass == 1){
    char *err = NULL;
    save_cfg(cfg, &err);
    free(err);
  }
}
```

JS code:
```
let Cfg_ = {
  // helper functions copied from api_config.js, enabling us to set config values
  setcfg: ffi('void save_config(void *, void *)'),
  set: function(path, setting){
    let entry = Cfg._find(path, Cfg._desc());
    if (entry === null) return undefined;
    if (setting !== Cfg.get(path)){
      this.setcfg(entry, setting);
    }
    let ev = 3;
    if (path.slice(0,6) !== 'nuset.'){
      ev = ev | 1<<3;
    }
    log_event(ev, path + ', ' + setting);
  },
};
```

closed 2017-09-19 by rojer