cesanta/mongoose-os #339: Slowdown/hand after configuration/init.js save

luseengithub, 2017-08-27:
I have got NodeMCU esp8266. The app which I am running was working very fine until today. 
Steps which I am doing:
- mos flash  - no issues
- mos ls  -no issues
- opening browser and connecting to the board without wifi setup - no issues
- selecting init.js and selecting Save and Reboot

After this device gets very slow, after restarting and connecting to the web if prints different characters, like memory dump.

luseengithub, 2017-08-28:
Looks like crash happens after saving the configuration:

`int disable_wifi_ap (void) {
  struct sys_config *cfg = get_cfg();
  cfg->wifi.ap.enable = false;
  char *err = NULL;
  save_cfg(cfg, &err); /* Writes conf9.json */
  printf("Saving configuration: %s\n", err ? err : "no error");
  free(err);
  return 0;
}`

What is wrong in this code?

cpq, 2017-09-13:
Please show your whole init.js

closed 2017-09-19 by cpq