cesanta/mongoose-os #342: aws-iot-setup no AWS China region(cn-north-1) support

elbertwang, 2017-08-29:
in mos/aws.go

func getAWSRegions() []string {
  	resolver := endpoints.DefaultResolver()
  	partitions := resolver.(endpoints.EnumPartitions).Partitions()
 	endpoints := partitions[0].Services()["iot"]
partition[0] only contain AWS global region

I made this change in my repo, and re-make mos.
https://github.com/cesanta/mongoose-os/pull/341/files\

still not work. must sth I didn't find.

elbertwang, 2017-08-29:
Finally, I find the issue. need update aws go sdk to latest to support China region.

govendor fetch github.com/aws/aws-sdk-go/aws
 govendor fetch github.com/aws/aws-sdk-go/aws/awserr
 govendor fetch github.com/aws/aws-sdk-go/aws/awsutil
 govendor fetch github.com/aws/aws-sdk-go/aws/client
 govendor fetch github.com/aws/aws-sdk-go/aws/client/metadata
 govendor fetch github.com/aws/aws-sdk-go/aws/corehandlers
 govendor fetch github.com/aws/aws-sdk-go/aws/credentials
 govendor fetch github.com/aws/aws-sdk-go/aws/credentials/ec2rolecreds
 govendor fetch github.com/aws/aws-sdk-go/aws/credentials/endpointcreds
 govendor fetch github.com/aws/aws-sdk-go/aws/credentials/stscreds
 govendor fetch github.com/aws/aws-sdk-go/aws/defaults
 govendor fetch github.com/aws/aws-sdk-go/aws/ec2metadata
 govendor fetch github.com/aws/aws-sdk-go/aws/endpoints
 govendor fetch github.com/aws/aws-sdk-go/aws/request
 govendor fetch github.com/aws/aws-sdk-go/aws/session
 govendor fetch github.com/aws/aws-sdk-go/aws/signer/v4
 govendor fetch github.com/aws/aws-sdk-go/private/protocol
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/json/jsonutil
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/jsonrpc
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/query
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/query/queryutil
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/rest
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/restjson
 govendor fetch github.com/aws/aws-sdk-go/private/protocol/xml/xmlutil
 govendor fetch github.com/aws/aws-sdk-go/service/iot
 govendor fetch github.com/aws/aws-sdk-go/service/govendor

elbertwang, 2017-08-30:
file pull request https://github.com/cesanta/mongoose-os/pull/341

cpq, 2017-09-13:
That got integrated, thank you

closed 2017-09-13 by cpq