cesanta/mongoose-os #343: HELP! WPA2 Enterprise Causes Core Dump on esp 8266

calkoe, 2017-08-30:
Everytime i try to connect my ESP8266 with a WPA2 Enterprise Secured Network the System Crashes and an the Core dump will be shown.
The Crash occurs sometime before the chip got an IP-Adress and sometimes after.

I tested a few App-Examples and played with the Configuration around a littebit but nothing helps.

calkoe, 2017-08-30:
`[Aug 30 15:45:58.774] EAP-PEAP: Phase 2 Request: type=26
[Aug 30 15:45:58.775] EAP-MSCHAPV2: RX identifier 95 mschapv2_id 94
[Aug 30 15:45:58.776] EAP-PEAP: received 53 bytes encrypted data for Phase 2
[Aug 30 15:45:58.783] EAP-PEAP: received Phase 2: code=1 identifier=96 length=11

[Aug 30 15:45:58.788] EAP-PEAP: Phase 2 Request: type=33
[Aug 30 15:45:58.882]
[Aug 30 15:45:58.882] connected with MY-NETWORK, channel 9
[Aug 30 15:45:58.883] dhcp client start...
[Aug 30 15:45:58.884] mgos_net_on_change_c WiFi STA: connected
[Aug 30 15:46:00.302] ip:172.25.20.13,mask:255.255.0.0,gw:172.25.1.1
[Aug 30 15:46:00.310] mgos_net_on_change_c WiFi STA: ready, IP 172.25.20.13, GW
172.25.1.1, DNS 172.25.2.1
[Aug 30 15:46:19.590] mgos_http_ev         0x3fff16cc HTTP connection from 172.2
5.42.69:63104
[Aug 30 15:46:19.621]
[Aug 30 15:46:19.621] Exception 28 @ 0x40106921, vaddr 0x0000051c
[Aug 30 15:46:19.623]  A0: 0x40106998  A1: 0x3ffffc70  A2: 0x0000051c  A3: 0x16d
00004
[Aug 30 15:46:19.631]  A4: 0x00000000  A5: 0x0fffffff  A6: 0x000016d0  A7: 0x000
00b68
[Aug 30 15:46:19.633]  A8: 0x0000fafa  A9: 0x00000b68 A10: 0x3fff1960 A11: 0x000
00002
[Aug 30 15:46:19.640] A12: 0x16d00004 A13: 0x00000362 A14: 0x40254a9b A15: 0x000
0001c
[Aug 30 15:46:19.648]
[Aug 30 15:46:19.648] (exc SP: 0x3ffffad0)
[Aug 30 15:46:19.649]
[Aug 30 15:46:19.649] --- BEGIN CORE DUMP ---
[Aug 30 15:46:19.650] {"arch": "ESP8266",
[Aug 30 15:46:19.650] "REGS": {"addr": 1073740544, "data": "
[Aug 30 15:46:19.658] mGkQQHD8/z8cBQAABADQFgAAAAD///8P0BYAAGgLAAD6+gAAaAsAAGAZ/z
8CAAAABADQFmIDAACbSiVAHAAAACFpEEAAAAAAAAAAAAAAAAAAAAAAMAAAAA==", "crc32": 380904
2835},
[Aug 30 15:46:19.667] "DRAM": {"addr": 1073643520, "data": "
[Aug 30 15:46:19.673] c84gQAAAAAAQgP4/AAAAAAIAAAD8gv4/ZIP+P8yD/j8AAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACEgyBAAQAAAKAhIUBkBf8/AAAAAAAAAAC8BP8/AAAAAAAAAA............................`

cpq, 2017-09-13:
Please provide a coredump backtrace.
See https://mongoose-os.com/docs/overview/debug.html#analysing-core-dumps

To get an .elf file for you firmware, download it https://mongoose-os.com/downloads/mos-ARCH-RELEASE.elf, e.g. https://mongoose-os.com/downloads/mos-esp8266-1.15.elf

cpq, 2017-09-26:
Closing, as no input

closed 2017-09-26 by cpq

calkoe, 2017-10-09:
For all out there with the same problem, i've found the following answer:

`ESP8266 has so little available RAM, and WPA-ENT uses so much of it, that there's hardly any left for the app anyway.
i'd say, for WPA-ENT support, go to ESP32, where it doesn't crash and which has enough RAM to actually do useful stuff while using WPA-ENT auth.`

_Thanks to_ `rojer` _on forum.mongoose-os.com_