cesanta/mongoose-os #347: Unable to flash app

OPatrick, 2017-09-04:
Hello,
Mongoose first install went ok. Can see files in devices. But flashing an example (Arduino basic usage) is not possible and led to this error:

Calling: Sys.GetInfo
[Sep  4 19:16:10.270] E0904 19:16:10.270144    5556 ui.go:513] Call complete, error: context deadline exceeded
[Sep  4 19:16:10.270] E0904 19:16:10.270144    5556 reconnect_wrapper.go:72] [reconnectWrapperCodec to COM5; connecting...] Connection closed

Same for any ap example. I did request "Rebuild App" before "Flash App firmware to device".
Flasher can not be stopped. Have to kill mos.exe (windows 10).
Thanks

benjamind, 2017-09-11:
I'm having the same issue, here's the output from the mos tool under windows 10 with verbose 4: https://gist.github.com/benjamind/5d7d574be8df9be870fcfdf12a532dec

cpq, 2017-09-13:
Could you confirm that there is no other mos.exe processes hanging around ?

OPatrick, 2017-09-13:
Thanks for your interest :-)
No other mos.exe listed by windows task manager.
Upgrading the tool recently did not make things better (mos-esp32-1.15 nor iot-kit-esp32-1.15)
Attached error log [error.txt](https://github.com/cesanta/mongoose-os/files/1298767/error.txt)

But...
I've seen this related subject:
https://forum.mongoose-os.com/discussion/1579/unable-to-finish-flashing-device-setup-with-webui-esp32
I did it "mos flash mos-esp32-1.14" and it kind of worked! (the flashing LED in the init.js example was not flashing, but I could see the device Logs in the IDE
Tried to re-flash with ide via "Device Setup" window but failed again with the above error message.
Trying "mos flash mos-esp32-1.15" is just making the command line no more responding.

cpq, 2017-09-19:
what board you're flashing to?
from the logs i see that the USB connection is quite flaky, and several attempts are required for success.

OPatrick, 2017-09-20:
Hello,
Well, I'm using this board: [TOGO](https://fr.aliexpress.com/item/TTGO-LORA-SX1278-ESP32-0-96OLED-16-Mt-bytes-128-Mt-bit-433Mhz-for-arduino/32824209385.html?spm=a2g0w.10010108.1000016.1.3f80957dwVKeNQ&isOrigTitle=true)
which is a clone of this one: [HelTec](https://fr.aliexpress.com/item/SX1278-LoRa-ESP32-0-96-inch-Blue-OLED-Display-Bluetooth-WIFI-Lora-Kit-32-Module-Internet/32822676113.html?spm=a2g0w.10010108.1000016.1.3f80957d1j1SFx&isOrigTitle=true)
But also on this board with same results: [Weemos](https://fr.aliexpress.com/item/WEMOS-WiFi-Bluetooth-Battery-ESP32-development-tool-ESP32-battery-esp8266-ESP-WROOM-32-ESP32/32819459737.html?spm=a2g0w.10010108.1000016.1.3f80957d7goWMu&isOrigTitle=true)
What is strange is that I never encountered any connection problem under Arduino IDE nor Platform IO...
Flashing is always a breeze...
Because I cannot use Mongoose-OS yet, I continue my work on Platform IO. And is goes well so far.
But I am desperate to use MOS ;-)
I also tried different ports on the PC (laptop ASUS ROG with windows 10 + 16 go RAM)
Thanks for your help.

closed 2017-09-20 by OPatrick

jafetmorales, 2017-10-04:
Was this problem fixed?

jafetmorales, 2017-10-04:
It was not the RPC, in my case. I just needed to press the Boot and the EN buttons in my board and leave them pressed both until the console that I am using attempted to connect to the ESP32 ROM for the 4th time. The flashing program said that everything was downloaded successfully, and the device happily replied "hello".

Anshu6943, 2018-02-09:
I have just updated mos tool to 1.26 but now I am unable to build programs