cesanta/mongoose-os #349: Is way to measure VCC in java?

aavestan, 2017-09-10:
I simply get it in Nodemcu with this "adc.readvdd33()"
I suggest you put it in "api_adc.js" to use simply
thanx

cpq, 2017-09-13:
@ruslanvaliullin please export native SDK functions for ESP8266 and ESP32 to read battery voltage.

smth like system_get_vdd33.

Should go to https://github.com/mongoose-os-libs/mjs/blob/master/esp8266/fs/api_esp8266.js , https://github.com/mongoose-os-libs/mjs/blob/master/esp32/fs/api_esp32.js

jocodeinc, 2017-09-16:
Couldn't this be done with ADC.read(pin) already?

ruslanvaliullin, 2017-09-19:
@aavestan, implemented https://github.com/mongoose-os-libs/adc/commit/f5412e8705415854790f33ce402c4de75bde802f

aavestan, 2017-09-25:
@ruslanvaliullin 
I check it in Wemos D1 mini and read ADC.read(0) :
It seems ADC(0) pin must be connect to VCC and it works fine.
Thanx

closed 2017-09-25 by ruslanvaliullin