cesanta/mongoose-os #350: c_web_config example doesn't work

bguban, 2017-09-11:
c_web_config example is broken.

PS examples are the best documentation. Try to keep them in a good state. Thanks.

cpq, 2017-09-11:
Could you be more specific please.
What exactly is broken, what browser you're using, etc.

bguban, 2017-09-12:
[image]
I've flashed the example to my esp8266 and went to it's web page. But there is only list of files instead of configuration page. I checked git log and found that index page was removed.

cpq, 2017-09-12:
Got it.

Don't use things from `fw/examples`, that's deprecated stuff.
https://mongoose-os.com/apps.html please. There is a web ui setup app.

closed 2017-09-12 by cpq