cesanta/mongoose-os #351: Add Flag to Remove Device Id from MQTT endpoint

applecrusher, 2017-09-12:
Hey All,

I am trying to remove the device id from MQTT endpoint, but it seems to get added automatically.

A reason to use this is because, for example, AT&T uses the following endpoint format:

> m2x/<M2X API Key>/requests

But filling it out in the mqtt section will always have it end with m2x/<M2X API Key>/requests/<id>

If this can be done another way, how?

cpq, 2017-09-13:
@applecrusher could you elaborate please? What exactly you're doing?

applecrusher, 2017-09-14:
I am trying to implement MQTT on the AT&T IOT platform, which doesn't end with the device id. 

[https://m2x.att.com/developer/documentation/v2/mqtt](https://m2x.att.com/developer/documentation/v2/mqtt)

cpq, 2017-09-19:
That's understood, thank you.

I still don't understand why do you think mOS appends anything to the MQTT topics.
If you're talking about disabling RPC over mqtt, do `rpc.mqtt.enable=false`.
Otherwise, mOS does not append anything to your MQTT topics.

applecrusher, 2017-09-28:
Ok. I got confused and thought the MQTT device address was the same as the MQTT server endpoint.

closed 2017-12-13 by dimonomid