cesanta/mongoose-os #353: 1.15 crashes on ESP32

benjamind, 2017-09-17:
I am unable to get the 1.15 or master builds of the mongoose-os firmware to work on the ESP32. The web interface always fails to connect properly and results in a log something like this:

```
Loaded https://mongoose-os.com/downloads/api.json
[Sep 16 17:52:21.886] E0916 17:52:21.886453   10198 ui.go:494] Calling: FS.ListExt
[Sep 16 17:52:21.888] E0916 17:52:21.888760   10198 ui.go:494] Calling: Sys.GetInfo
Server Error: Lost connection with your board. Please restart mos tool from terminal
[Sep 16 17:52:23.952] E0916 17:52:23.951961   10198 ui.go:518] Call complete, error: context deadline exceeded
[Sep 16 17:52:23.955] E0916 17:52:23.954910   10198 reconnect_wrapper.go:72] [reconnectWrapperCodec to /dev/ttyUSB1; connecting...] Connection closed
[Sep 16 17:52:25.743] E0916 17:52:25.743486   10198 ui.go:494] Calling: FS.ListExt
[Sep 16 17:52:27.026] E0916 17:52:27.025991   10198 ui.go:518] Call complete, error: context deadline exceeded
[Sep 16 17:52:27.030] E0916 17:52:27.029600   10198 reconnect_wrapper.go:72] [reconnectWrapperCodec to /dev/ttyUSB1; connecting...] Connection closed
Server Error: Lost connection with your board. Please restart mos tool from terminal
[Sep 16 17:52:27.833] E0916 17:52:27.833050   10198 ui.go:518] Call complete, error: context deadline exceeded
[Sep 16 17:52:27.833] E0916 17:52:27.833174   10198 reconnect_wrapper.go:72] [reconnectWrapperCodec to /dev/ttyUSB1; connect in 1.20s] Connection closed
[Sep 16 17:52:30.079] E0916 17:52:30.079843   10198 ui.go:494] Calling: FS.ListExt
[Sep 16 17:52:30.084] E0916 17:52:30.084182   10198 ui.go:494] Calling: Sys.GetInfo
```

While the device log outputs random gibberish.

I would like to compile the 1.14 version of the firmware from local and use the 1.14 compatible libs and modules, but I cannot seem to figure out how to do this. I tried setting `libs_version` and `modules_version` and `mongoose_os_version` in the `mos.yml` but when I try to compile against the 1.14 tag of mongoose-os I get a whole load of undefined reference errors and it still appears to try and compile against 1.15 libs.

Any suggestions?

cpq, 2017-09-17:
mos update 1.14

benjamind, 2017-09-17:
doesn't appear to work, when I run that on ubuntu 16.04 I just end up with 1.15 again. I installed mos through the apt repository.

In further investigation it appears this is just a serial issue. The application is still running, but the console output is no longer being parsed correctly at 115200 baud. I've tried a bunch of other baud rates to no effect, perhaps some other setting is not being setup correctly in 1.15?

benjamind, 2017-09-17:
Baud rate 76923 actually works: https://www.esp32.com/viewtopic.php?t=2043

cpq, 2017-09-17:
I am confused now, so is that a 1.15 introduced an issue? Did 1.14 work?

benjamind, 2017-09-17:
Yes 1.15 introduced this issue. If i manually flash the precompiled 1.14 binary then the output is correct and everything is fine. If I flash 1.15 I get garbage on the console until running with `--baud-rate 76923`.

rojer, 2017-09-18:
i can confirm the issue, will investigate in the near future.

sh4nnongoh, 2017-09-22:
Same issue here

guywithaview, 2017-09-26:
I am also seeing this issue!

closed 2017-09-26 by rojer