cesanta/mongoose-os #354: FR: Add GPIO output open drain mode

nliviu, 2017-09-18:
Sometimes it is necessary to use a GPIO in open drain mode which currently is not supported in Mongoose OS.

j0ker77, 2018-01-02:
I had the same requirement to use a GPIO of my ESP8266 in opendrain mode with MJS.
I extended main.c by my own my_mgos_gpio_set_mode and ffied it in my_gpio,js.
I also wrote a version for ESP32 but did not test yet, let me know if you are interested in it.

```
// main.c
// -------------------------------- 
// extend gpio mode by opendrain for esp8266                      
// -------------------------------- 
//
// mgos_gpio_set_mode copied from https://github.com/cesanta/mongoose-os/blob/master/fw/platforms/esp8266/src/esp_gpio.c
// opendrain mode copied from nodemcu's https://github.com/nodemcu/nodemcu-firmware/blob/master/app/platform/platform.c
//

#include "fw/platforms/esp8266/src/esp_gpio.h"

#ifdef RTOS_SDK
#include <esp_common.h>
#include <driver_lib/include/gpio.h>
#else
#include <user_interface.h>
#endif

#include "mgos_bitbang.h"
#include "mgos_gpio.h"
#include "mgos_gpio_hal.h"

#include "common/platforms/esp8266/esp_missing_includes.h"
#include "common/cs_dbg.h"
#include "esp_periph.h"

void gpio16_output_conf(void);
void gpio16_input_conf(void);

enum my_mgos_gpio_mode {
  MY_MGOS_GPIO_MODE_INPUT = 0, /* input mode */
  MY_MGOS_GPIO_MODE_OUTPUT = 1, /* output mode */
  MY_MGOS_GPIO_MODE_OPENDRAIN = 2 /* opendrain mode */
};

IRAM bool my_mgos_gpio_set_mode(int pin, enum my_mgos_gpio_mode mode) {
  if (pin == 16) {
    if (mode == MY_MGOS_GPIO_MODE_INPUT) {
      gpio16_input_conf();
    } else {
      gpio16_output_conf();
    }
    return true;
  }

  if (pin >= 6 && pin <= 11) {
    LOG(LL_ERROR, ("GPIO%d is used by SPI flash, don't use it", pin));
    /*
     * Alright, so you're here to investigate what's up with this error. So,
     * GPIO6-11 are used for SPI flash and messing with them causes crashes.
     * In theory, if flash is configured for double or single mode, GPIO9 and
     * GPIO10 can be reclaimed. And /CS could be hard-wired to 0 if flash is
     * the only device, reclaiming GPIO11. However, on all the modules currently
     * in existence they are connected to the flash chip's pins, which, when not
     * used for quad i/o, are usually /HOLD and /WP. Thus, using them will still
     * crash most ESP8266 modules, but in a different way.
     * So really, just stay away from GPIO6-11 if you can help it.
     * If you are sure you know what you're doing, copy the code below.
     */
    return false;
  }

  const struct gpio_info *gi = get_gpio_info(pin);
  if (gi == NULL) return false;

  switch (mode) {
    case MY_MGOS_GPIO_MODE_INPUT:
      PIN_FUNC_SELECT(gi->periph, gi->func);
      GPIO_REG_WRITE(GPIO_ENABLE_W1TC_ADDRESS, BIT(pin));
      break;
    case MY_MGOS_GPIO_MODE_OUTPUT:
      PIN_FUNC_SELECT(gi->periph, gi->func);
      GPIO_REG_WRITE(GPIO_ENABLE_W1TS_ADDRESS, BIT(pin));
      gpio_pin_intr_state_set(GPIO_ID_PIN(pin), GPIO_PIN_INTR_DISABLE);
      GPIO_REG_WRITE(GPIO_STATUS_W1TC_ADDRESS, BIT(pin));
      /* Pin driver -> direct drive, pad driver -> open drain. */
      GPIO_REG_WRITE(GPIO_PIN_ADDR(GPIO_ID_PIN(pin)),
                     GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(pin))) &
                         (~GPIO_PIN_PAD_DRIVER_SET(GPIO_PAD_DRIVER_DISABLE)));
      break;
    case MY_MGOS_GPIO_MODE_OPENDRAIN:
      PIN_FUNC_SELECT(gi->periph, gi->func);
      GPIO_REG_WRITE(GPIO_ENABLE_W1TS_ADDRESS, BIT(pin));
      gpio_pin_intr_state_set(GPIO_ID_PIN(pin), GPIO_PIN_INTR_DISABLE);
      GPIO_REG_WRITE(GPIO_STATUS_W1TC_ADDRESS, BIT(pin));
      /* Pin driver -> direct drive, pad driver -> open drain. */
      GPIO_REG_WRITE(GPIO_PIN_ADDR(GPIO_ID_PIN(pin)),
                     GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(pin))) |
                         GPIO_PIN_PAD_DRIVER_SET(GPIO_PAD_DRIVER_ENABLE));   // copied from nodemcu "platform.c"
      break;
    default:
      return false;
  }

  return true;
}
```

```
//----------------------------------------------------------------------
// my_gpio.js

load('api_gpio.js');

// extend set_mode with OPENDRAIN
let myGPIO = {
    set_mode: ffi('int my_mgos_gpio_set_mode(int,int)'),
    MODE_INPUT: 0,
    MODE_OUTPUT: 1,
    MODE_OPENDRAIN: 2
  };
```

Usage:
```
load('my_gpio.js');
let pin = 14;
myGPIO.set_mode(pin, myGPIO.MODE_OPENDRAIN);
GPIO.set_pull(pin, GPIO.PULL_NONE);
GPIO.write(pin, 0);  // or 1
```

nliviu, 2018-01-04:
I needed it for ESP8266 and did this:
```
mgos_gpio_set_mode(_pinOut, MGOS_GPIO_MODE_OUTPUT);
#if CS_PLATFORM == CS_P_ESP8266
        GPIO_REG_WRITE(GPIO_PIN_ADDR(GPIO_ID_PIN(_pinOut)), GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(_pinOut))) | (GPIO_PIN_PAD_DRIVER_SET(GPIO_PAD_DRIVER_ENABLE)));
#endif
```

nliviu, 2018-01-27:
@rojer Thank you!

closed 2018-01-27 by nliviu