cesanta/mongoose-os #355: Deep sleep functionality added for CC3200

conniercarey, 2017-09-22:
It would be beneficial to add deep sleep functionality for the CC3200 specifically the low power deep sleep.  Power dependent implementations that require large amounts of power would benefit greatly from such a feature.  I have tried to bring in the sleep framework from TI directly but hit many roadblocks with the Mongoose integration.  This feature would benefit many who are using mongoose os on their products.

kzyapkov, 2017-09-22:
+1

We tried this as well, but there were too many quirks and roadblocks.

cpq, 2017-09-25:
It would help a lot if you describe those roadblocks - could you put together a description please?

Another question is that whether CC3200 is a perspective platform at all. We'd rather add this to CC3220.

conniercarey, 2017-09-25:
In order to do a low power deep sleep on the CC3200, code from the ti middleware would need to be incorporated into the OS. I have tried to bring over the middleware just locally to my project but still run into problems which I believe could possibly be due to the initialization sequence of the device itself and/or the way it is configured.   In seeking support from ti in an attempt to get this functionality to work, they told me to use their example verbatim but that is not possible when using Mongoose OS.  Also, it doesn't make sense for me to carry the middleware in my image.  It would be beneficial to have this functionality incorporated into the Mongoose OS support of the CC3200.  Without it, low power deep sleep functionality that the chipset has is not available to users of the CC3200 and this functionality is extremely beneficial.  Please consider adding it to the CC3200 device.  Thank you!

cpq, 2017-09-26:
Thanks.
Your description does not shed any light on what actual roadblocks there are.

We'll consider adding this to the CC3220.
Investing into CC3200 without any commercial engagement is not feasible for us.

closed 2017-09-26 by cpq

conniercarey, 2017-09-26:
Okay, I'm not sure how you can say you support a product such as CC3200 and not include all the functionality.  Your solution does not include the middleware needed to perform a Low Power Deep Sleep.   The Mongoose OS solution does not provide a solution to perform low power deep sleep which is a crucial function to all users.  So are you no longer supporting CC3200?  What you have now is all that you will support?  Even though there are pieces and functionality missing?  When you list that Mongoose OS supports the CC3200 chipset make sure you also list the functionality you don't provide so that user can make an educated decision on whether or not to use your OS.

cpq, 2017-09-26:
You are having a wrong impression of what Mongoose OS is, and what it is supposed to do.