cesanta/mongoose-os #359: error response: 400: max size 2097152 exceeded (upload size is 10037871)

wangwangwang1998, 2017-09-30:
Hi,    
I create a new project, and run `mos build --arch esp32` and I got the following error message:    

`Uploading sources (10038588 bytes)`  
`Error: error response: 400: max size 2097152 exceeded (upload size is 10037871)`  

I want to know if there is any ways to solve it.  

Thank you!

calkoe, 2017-10-09:
Hi wangwangwang1998,

The size of the filesystem is limited to round about 200kb. That means that you have to make sure that the size of the /fs folder in your Mongoose-OS Project is smaller than 200kb (note that some config files + mjs files already use 132 of your Space).

**+1** 
I've got the same Problem.. i need more space for HTTP files in the fs. Is there any way to increase the fs size ? i've tried `mos build --arch esp8266 --build-cmd-extra FLASH_SIZE = 4M --build-cmd-extra FS_SIZE = 1M` already but that didn't work..

cpq, 2017-10-11:
@wangwangwang1998 

2Mb is the sensible limit for the amount of sources sent to the build machine. The build command zips your app directory - so make sure you don't have any log files or another junk in the app directory.

closed 2017-10-11 by cpq