cesanta/mongoose-os #360: Loading Mongoose firmware into Sonoff devices via SonOTA?

PKGeorgiev, 2017-10-05:
Recently I stumbled across this: [https://github.com/mirko/SonOTA](url)
I'm curious if it is possible to flash sonoffs with Mongoose OS via this OTA machanism?

cpq, 2017-10-12:
Mongoose OS provides its own OTA mechanism:
https://mongoose-os.com/docs/book/ota.html

PKGeorgiev, 2017-10-12:
I know that. The idea is to skip initial need for cables to deploy Mongoose OS to sonoff devices.

cpq, 2017-10-12:
Ah. Could you elaborate more please?

PKGeorgiev, 2017-10-12:
Thanks. 
Sonoff devices come pre-loaded with proprietary firmware by Itead. Their firmware support OTA but officially it cannot be used by the end users. 

So the only "official" way to flash newer sonoff device is the classic one: disassemble it, use soldering iron, solder a wire go GPIO0 etc. Not very hard, but is time consuming.

Recently one guy had discovered the internals of Itead's OTA mechanism: http://blog.nanl.de/2017/05/sonota-flashing-itead-sonoff-devices-via-original-ota-mechanism

Then another programmer made a Python script that simulates Itead's OTA servers making it possible to re-flash sonoff devices locally without taking them apart (i.e. taking advantage of the original OTA mechanism): https://github.com/mirko/SonOTA

This library is used to load firmware based on Tasmota:
 https://github.com/mirko/SonOTA/tree/master/static. But my firmware knowledge is very limited.

There are a few steps described in SonOTA's readme.

My question is that - will it be possible to use SonOTA (or similar script, eventually written by you) to re-flash sonoff devices with Mongoose OS firmware.  This will allow quick and easy way of initial "Mongoose-ing" any sonoff device via existing OTA mechanism. And of course this will definitely increase Mongoose OS' popularity :)

dimonomid, 2017-12-13:
I think it's possible, but not a priority at the moment.