cesanta/mongoose-os #363: {"error":"failed toUnable to add library to a project: git ls-files: fatal: Not a git repository (or any of the parent directories): .git\n: exit status 128"}

jafetmorales, 2017-10-09:


operationflash, 2017-10-26:
I'm having the same problem for getting the adc library for a ESP32

jafetmorales, 2017-10-26:
You might need to go to your .mos directory and delete the adc library folder from there, if you added it and tried to delete it before.

operationflash, 2017-10-27:
I removed the adc library from the .mos directory and this didn't help.
After that I manually added the adc library folder to .mos and and tried to build it with command line:
`sudo mos build --arch esp32`
`Connecting to https://mongoose.cloud, user test`
`Uploading sources (1911 bytes)`
`Error: error response: 400: container exited with 1 status code`

jafetmorales, 2017-10-27:
I remember having that same error as well, and I concluded (although not 100% sure at this point) that it was a temporary problem with their remote building service (you're uploading the source and building it in their servers). Their servers may have had a docker error and sent the error message back to you. I just tried myself building right now and it worked, so it might be working again. However, if the problem persists, I would try to build locally in your machine. For that you would need to download docker and then execute:

`mos build --arch esp32 --local --verbose`

Good luck.

farooqhasny, 2017-11-15:
manually deleting the app from C:\Users\user\.mos\apps-1.19\ did fix this problem for me

operationflash, 2017-11-15:
Hey thanks for the responses. I had flashed the ESP32 with the GCP image, which had no ADC included. But the normal firmware version had the ADC build in, so I flashed this firmware and added the necessary GCP code. 

@jafetmorales building it locally would have been a good option.
@farooqhasny forgot to tell that I was using Linux.

dimonomid, 2017-12-13:
I think it's irrelevant now, because all libs are now located under `deps` dir in the app's directory.

jafetmorales, 2017-12-13:
@dimonomid so you mean that in the latest version of the Web UI you don't need to manually go and delete the file? If am not mistaken, you are saying that before the dependency (i.e., api_wifi.js) was being stored in .mos, but not in the app's directory, and now it is stored in the app's directory. Is that what you are saying?

dimonomid, 2017-12-13:
Let me elaborate a bit.

In `mos.yml` we can have some deps, e.g.:
```
libs:
  - origin: https://github.com/mongoose-os-libs/dht
  - origin: https://github.com/mongoose-os-libs/js-demo-bundle
```
So previously all those deps were stored under `~/.mos/libs-<mos.version>`: given the snippet above, there would be `~/.mos/libs-1.19/dht-1.19`, `~/.mos/libs-1.19/js-demo-bundle-1.19`, and all the deps of deps.

Right now, when one does a "remote build" (and Web UI can only do remote builds atm), nothing will be stored anywhere. Neither under `~/.mos/libs-...`, nor under app's dir.

When one does a "local build" (in the simplest case, it's `mos build --local`), all the deps will be cloned under `deps` dir in the app's directory.

closed 2017-12-14 by dimonomid