cesanta/mongoose-os #365: [enhancement] Configuration handling

guglielmino, 2017-10-14:
In the last version configuration is no longer exposed by a struct but a getter and a setter function for each field in the configuration itself (as described [here](https://github.com/cesanta/mongoose-os/commit/03a413b9ee284ae8fad2dba70b3383fc45eced04)).

I think it could be a better approach to use per data type getter/setter with the filed key as string.
So, in current version accessing to update_timeout, for example, is `mgos_sys_config_get_update_timeout()`
while it should become `mgos_sys_config_get("update_timeout")` and the same for other data types. 

Using static functions gives build time check but isn't good if you need to access to config fields dynamically.

dimonomid, 2017-12-13:
Build time check is a huge benefit, so we believe static functions should be the preferred way to access config values; whereas accessing config fields dynamically looks like sort of a corner case. I do agree that supporting that corner case as well would be good, but it's not a priority at the moment.

That said, *reading* config values by paths like `"mqtt.enable"` is possible already: you can get a config schema entry by its path with [mgos_conf_find_schema_entry](https://mongoose-os.com/docs/api/mgos_config_util.h.html#*mgos_conf_find_schema_entry) (so something like `mgos_conf_find_schema_entry("mqtt.enable", mgos_config_schema())`), and then get its value using `mgos_conf_value_.....`, see https://mongoose-os.com/docs/api/mgos_config_util.h.html#*mgos_conf_value_string .

dimonomid, 2017-12-14:
By the way, be aware that we can use this:
```c
mgos_config_apply("{\"wifi\": {\"ap\": {\"enable\": false}}}", true);
```
For such a corner case this may actually be enough, what do you think?

guglielmino, 2017-12-31:
I didn't know about this opportunity, it's fine even if writing JSON in a C string is a bit error prone.

dimonomid, 2017-12-31:
Yeah JSON in a C string is not only a bit error prone but also notably ugly, but for such a corner case, meh. :)
Ok cool, closing this one then.

closed 2017-12-31 by dimonomid