cesanta/mongoose-os #366: mos tool over MQTT

SBartley08, 2017-10-14:
Hi there 

I am ruining the latest mos tool on windows 64bit and works fine over Serial but fails to connect with MQTT, console shows :

Loaded https://mongoose-os.com/downloads/api.json
[Oct 14 15:08:35.374] E1014 15:08:35.374133    2420 ui.go:507] Calling: Sys.GetInfo
[Oct 14 15:08:38.382] E1014 15:08:38.382691    2420 ui.go:531] Call complete, error: context deadline exceeded
[Oct 14 15:08:40.861] E1014 15:08:40.860536    2420 ui.go:507] Calling: FS.ListExt
[Oct 14 15:08:43.863] E1014 15:08:43.863305    2420 ui.go:531] Call complete, error: context deadline exceeded
Server Error: Lost connection with your board. Please restart mos tool from terminal

however I can see the log from the mqtt server : ( Topic | Payload)

devices/14017098/rpc | {"src":"mos-1507990227","id":1412520010320,"method":"Sys.GetInfo"}
-- | --
devices/14017098/rpc | {"id":1412520010320,"src":"14017098","dst":"mos-1507990227","result":{"app": "RGBLights", "fw_version": "1.0", "fw_id": "20171014-121349/???", "mac": "5ECF7FD5E24A", "arch": "esp8266", "uptime": 767, "ram_size": 52432, "ram_free": 36324, "ram_min_free": 31672, "fs_size": 233681, "fs_free": 130269,"wifi": {"sta_ip": "192.168.88.211", "ap_ip": "", "status": "got ip", "ssid": "XXXXXX"}}}

same issue when i try mos --port MQTT://... ls , I see the responce from the esp but never updates command line 
my config.json is below (with some details replaced with XXXXXX eg passwords ): 
{
  "sntp": {
    "enable": true,
    "server": "pool.ntp.org",
    "retry_min": 1,
    "retry_max": 30,
    "update_interval": 7200
  },
  "device": {
    "id": "14017098",
    "password": ""
  },
  "debug": {
    "udp_log_addr": "",
    "mbedtls_level": 0,
    "level": 2,
    "filter": "",
    "stdout_uart": 0,
    "stderr_uart": 0,
    "factory_reset_gpio": -1,
    "mg_mgr_hexdump_file": "",
    "stdout_topic": "",
    "stderr_topic": ""
  },
  "sys": {
    "mount": {
      "path": "",
      "dev_type": "",
      "dev_opts": "",
      "fs_type": "",
      "fs_opts": ""
    },
    "wdt_timeout": 30
  },
  "conf_acl": "*",
  "i2c": {
    "enable": false,
    "freq": 100000,
    "debug": false,
    "sda_gpio": 12,
    "scl_gpio": 14
  },
  "mjs": {
    "generate_jsc": true
  },
  "mqtt": {
    "enable": true,
    "server": "XXXXXcloudmqtt.com:18133",
    "client_id": "14017098",
    "user": "14017098",
    "pass": "XXXXXXXX",
    "reconnect_timeout_min": 2,
    "reconnect_timeout_max": 60,
    "ssl_cert": "",
    "ssl_key": "",
    "ssl_ca_cert": "",
    "ssl_cipher_suites": "",
    "ssl_psk_identity": "",
    "ssl_psk_key": "",
    "clean_session": true,
    "keep_alive": 60,
    "will_topic": "",
    "will_message": ""
  },
  "rpc": {
    "enable": true,
    "max_frame_size": 4096,
    "max_queue_length": 25,
    "default_out_channel_idle_close_timeout": 10,
    "acl_file": "",
    "auth_domain": "",
    "auth_file": "",
    "ws": {
      "enable": true,
      "server_address": "",
      "reconnect_interval_min": 1,
      "reconnect_interval_max": 60,
      "ssl_server_name": "",
      "ssl_ca_file": "",
      "ssl_client_cert_file": ""
    },
    "mqtt": {
      "enable": true,
      "topic": "devices/14017098/rpc",
      "is_trusted": true
    },
    "uart": {
      "uart_no": 0,
      "baud_rate": 115200,
      "fc_type": 2,
      "wait_for_start_frame": true
    }
  },
  "spi": {
    "enable": false,
    "debug": false,
    "miso_gpio": 12,
    "mosi_gpio": 13,
    "sclk_gpio": 14,
    "cs0_gpio": 5,
    "cs1_gpio": -1,
    "cs2_gpio": -1
  },
  "wifi": {
    "sta": {
      "enable": true,
      "ssid": "XXXXXXXX",
      "pass": "XXXXXXXX",
      "user": "",
      "anon_identity": "",
      "cert": "",
      "key": "",
      "ca_cert": "",
      "ip": "",
      "netmask": "",
      "gw": "",
      "nameserver": "",
      "dhcp_hostname": ""
    },
    "ap": {
      "enable": false,
      "ssid": "Mongoose_??????",
      "pass": "Mongoose",
      "hidden": false,
      "channel": 6,
      "max_connections": 10,
      "ip": "192.168.4.1",
      "netmask": "255.255.255.0",
      "gw": "192.168.4.1",
      "dhcp_start": "192.168.4.2",
      "dhcp_end": "192.168.4.100",
      "trigger_on_gpio": -1,
      "keep_enabled": true
    }
  },
  "neoPixel": {
    "pin": 2,
    "bufferSize": 5
  }
}

gojimmypi, 2017-10-24:
You have MQTT (mostly) working? (well, at least code complete compiling) cool. I was wondering where to find mgos_mqtt.h included here:

[https://github.com/cesanta/mongoose-os/blob/master/fw/examples/c_mqtt/src/main.c](
https://github.com/cesanta/mongoose-os/blob/master/fw/examples/c_mqtt/src/main.c)

The file does not seem to be anywhere in this repository and [https://github.com/mongoose-os/](https://github.com/mongoose-os/) is private.

Thanks

moosilauke18, 2017-10-26:
@gojimmypi here is the `mgos_mqtt.h` file: https://github.com/mongoose-os-libs/mqtt/blob/master/include/mgos_mqtt.h

dimonomid, 2017-12-13:
Failed to reproduce, here's what I'm doing:

```
$ mos flash esp8266
$ mos config-set wifi.ap.enable=false wifi.sta.enable=true wifi.sta.ssid=MYSSID wifi.sta.pass=MYPASS mqtt.enable=true mqtt.server=iot.eclipse.org:1883
$ mos --port mqtt://iot.eclipse.org:1883/esp8266_DA7E15 call Config.Get
```

dimonomid, 2017-12-14:
Closing this for now, feel free to reopen if it's still an issue for you

closed 2017-12-14 by dimonomid