cesanta/mongoose-os #367: what is the diff between iot-kit-esp8266-1.18 and mos-esp8266-1.18

kalsolio, 2017-10-18:
what is the diff between iot-kit-esp8266 and mos-esp8266

shantanuthatte, 2017-10-27:
Have the same question with esp32. `iot-kit-espxxxx-1.18 seems to have an empty `init.js`. Where as `mos-espxxxx-1.18` has a example `init.js` with blink and MQTT. What are the use-cases for this each?

digitalcraft, 2017-10-31:
a bump on this?
not sure why it takes so long to get an answer on this one.

gotfredsen, 2017-11-05:
another bump...

gotfredsen, 2017-11-05:
After a bit of comparing I noticed that the **mos-** is a more "capable" version than the **iot-kit-**. 2 things I noticed is that:
1. the **mos-** version have I2C RPC calls, the **iot-kit-** doesn't have that, and
1. the device config in the **mos-** version is having way more options.

My guess is that the **iot-kit-** version is a limited version of **mos-** to save memory and/or RTOS overhead.

_Bare in mind - I am guessing here, I would still like an official answer..._

shantanuthatte, 2017-11-05:
I do think the same, the **iot-kit** is a _lighter_ version. Running the same sample code in **mos** v/s **iot-kit** results in higher memory usage for the **mos** version. I also do notice the device getting slightly warmer on **mos**.  Will try to get more data on this.

dimonomid, 2017-12-13:
`mos-esp8266-X.XX` is an old name for the prebuilt demo-js app: https://github.com/mongoose-os-apps/demo-js. And iot-kit is another app: https://github.com/mongoose-os-apps/iot-kit.

They are similar, but iot-kit is a bit tailored for Google IoT Core IoT Kit; as opposed to demo-js it includes the dht lib and also doesn't contain demo led-blinking code and configuration values.

closed 2017-12-13 by dimonomid

dimonomid, 2017-12-13:
Sorry for not replying earlier, and now it's not quite relevant since iot-kit will be removed soon.