cesanta/mongoose-os #371: ESP32, WIFI constantly fails to connect and returns 201

sh4nnongoh, 2017-10-27:
Hi,

Wifi fails on ESP32 but works with ESP8266 with the exact same code.
On the ESP32 it constantly returns `Disconnected from XXX, reason: 201`.

This issues occures with the latest mos build and esp32 build.

xlfe, 2017-11-02:
I'm having this same issue

precobos, 2017-11-21:
Me too, after a reset connects without problems

kzyapkov, 2017-12-13:
Additionally, if a hardware timer is running, during WiFi reconnect the module crashes one out of two times. Once this happens it never recovers -- boots, crashes, dumps a core and repeats:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5412
load:0x40078000,len:0
load:0x40078000,len:14416
entry 0x400790bc
[0;32mI (69) boot: Detected ESP32[0m
[0;32mI (32) boot: ESP-IDF v1.0-2984-gc13c31f 2nd stage bootloader[0m
[0;32mI (32) boot: compile time 23:42:57[0m
[0;32mI (32) boot: Enabling RNG early entropy source...[0m
[0;32mI (38) boot: SPI Speed      : 40MHz[0m
[0;32mI (42) boot: SPI Mode       : DIO[0m
[0;32mI (46) boot: SPI Flash Size : 4MB[0m
[0;32mI (50) boot: Partition Table:[0m
[0;32mI (53) boot: ## Label            Usage          Type ST Offset   Length   Flags[0m
[0;32mI (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000 00000000[0m
[0;32mI (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000 00000000[0m
[0;32mI (78) boot:  2 app_0            OTA app          00 10 00010000 00180000 00000000[0m
[0;32mI (86) boot:  3 fs_0             SPIFFS           01 82 00190000 00040000 00000000[0m
[0;32mI (94) boot:  4 app_1            OTA app          00 11 001d0000 00180000 00000000[0m
[0;32mI (103) boot:  5 fs_1             SPIFFS           01 82 00350000 00040000 00000000[0m
[0;32mI (111) boot: End of partition table[0m
[0;32mI (115) boot: OTA data 0: seq 0x00000001, st 0x10, CRC 0x157a2b85, valid? 1[0m
[0;32mI (123) boot: OTA data 1: seq 0x00000000, st 0x00, CRC 0x00000000, valid? 0[0m
[0;32mI (131) boot: Disabling RNG early entropy source...[0m
[0;32mI (136) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x20a78 (133752) map[0m
[0;32mI (192) esp_image: segment 1: paddr=0x00030aa0 vaddr=0x3ffb0000 size=0x02754 ( 10068) load[0m
[0;32mI (196) esp_image: segment 2: paddr=0x000331fc vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (199) esp_image: segment 3: paddr=0x00033604 vaddr=0x40080400 size=0x0ca0c ( 51724) load[0m
[0;32mI (228) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x95fd8 (614360) map[0m
[0;32mI (444) esp_image: segment 5: paddr=0x000d5ff8 vaddr=0x4008ce0c size=0x034c4 ( 13508) load[0m
[0;32mI (449) esp_image: segment 6: paddr=0x000d94c4 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (460) cpu_start: Pro cpu up.[0m
[0;32mI (460) cpu_start: Single core mode[0m
[0;32mI (460) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (465) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (471) heap_init: At 3FFB8DB8 len 00027248 (156 KiB): DRAM[0m
[0;32mI (478) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (484) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (490) heap_init: At 400902D0 len 0000FD30 (63 KiB): IRAM[0m
[0;32mI (497) cpu_start: Pro cpu start user code[0m
[0;32mI (179) cpu_start: Starting scheduler on PRO CPU.[0m
mg_lwip_if_init      0x3ffc0b84 Mongoose init
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060031  A0      : 0x40081ced  A1      : 0x3ffb07f0  
A2      : 0x3ffb6378  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00060023  A7      : 0x0001f1f0  A8      : 0x800833f7  A9      : 0x3ffb07d0  
A10     : 0x00000000  A11     : 0x3ffdfff9  A12     : 0x80086f5e  A13     : 0x3ffc0660  
A14     : 0x3ffb8db8  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000 0x40081cea 0x4000bfed
--- BEGIN CORE DUMP ---
...
```

Happens with https://github.com/kzyapkov/beninca-wifi

mike-rogers, 2018-02-21:
According to [the source](https://github.com/cesanta/esp-idf/blob/master/components/esp32/include/esp_wifi_types.h#L91) a `201` indicates the AP can't be found. Does that help?

rojer, 2018-04-20:
i'm going to close this - i think it was fixed upstream since then.

closed 2018-04-20 by rojer