cesanta/mongoose-os #374: Can't build example c_mqtt for ESP32 board

jptalledo10, 2017-10-28:
Hello
I clone the repository into my Linux VM. I am able to run mos and flash the default firmware. When I try to compile the c_mqtt example, I am getting build errors. I am using mos build --arch esp32.  do I need to setup something before that?

the output is the following:

Connecting to https://mongoose.cloud, user test
Uploading sources (3634 bytes)
Handling lib "ca-bundle"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/ca-bundle-1.18"
Handling lib "http-server"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/http-server-1.18"
Handling lib "atca"...
Optional, skipping
Handling lib "ethernet"...
Optional, skipping
Handling lib "wifi"...
Optional, skipping
Handling lib "ota-http-server"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/ota-http-server-1.18"
Handling lib "http-server"...
Already handled, skipping
Handling lib "ota-http-client"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/ota-http-client-1.18"
Handling lib "rpc-loopback"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-loopback-1.18"
Handling lib "rpc-common"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-common-1.18"
Handling lib "http-server"...
Already handled, skipping
Handling lib "rpc-mqtt"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-mqtt-1.18"
Handling lib "mqtt"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/mqtt-1.18"
Handling lib "rpc-common"...
Already handled, skipping
Handling lib "rpc-service-config"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-service-config-1.18"
Handling lib "rpc-common"...
Already handled, skipping
Handling lib "rpc-service-fs"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-service-fs-1.18"
Handling lib "rpc-common"...
Already handled, skipping
Handling lib "rpc-service-gpio"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-service-gpio-1.18"
Handling lib "rpc-common"...
Already handled, skipping
Handling lib "rpc-service-i2c"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-service-i2c-1.18"
Handling lib "i2c"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/i2c-1.18"
Handling lib "rpc-common"...
Already handled, skipping
Handling lib "rpc-service-ota"...
The --lib flag was not given for it, checking repository
Prepared local dir: "/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/libs/rpc-service-ota-1.18"
Handling lib "ota-http-client"...
[... 217 lines cut ...]
/mongoose-os/common/cs_dbg.h:61:67: note: in definition of macro 'LOG'
     if (cs_log_print_prefix(l, __func__, __FILE__)) cs_log_printf x; \
                                                                   ^
/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/src/main.c: In function 'ev_handler':
/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/src/main.c:71:9: error: implicit declaration of function 'mgos_sys_config_get_mqtt_sub' [-Werror=implicit-function-declaration]
     if (mgos_sys_config_get_mqtt_sub() == NULL ||
         ^
/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/src/main.c:71:40: error: comparison between pointer and integer [-Werror]
     if (mgos_sys_config_get_mqtt_sub() == NULL ||
                                        ^
/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/src/main.c:72:40: error: comparison between pointer and integer [-Werror]
         mgos_sys_config_get_mqtt_pub() == NULL) {
                                        ^
cc1: all warnings being treated as errors
/mongoose-os/fw/platforms/esp32/src/esp32_src.mk:127: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/fwbuild-volumes/1.18/apps/c_mqtt/esp32/build_contexts/build_ctx_798599785/build/objs/mosapp'
/opt/Espressif/esp-idf/make/project.mk:421: recipe for target 'component-mosapp-build' failed
make: Leaving directory '/app'
make: *** [component-mosapp-build] Error 2
Error: exit status 2
Error: build failed

Any ideas what could be wrong?

cpq, 2017-10-28:
Please `mos update latest` and retry.

jptalledo10, 2017-10-28:
Thanks. It worked

closed 2017-10-28 by cpq

hqawasmi, 2018-02-13:
@cpq , above solution didn§t work for me , I am using Mac , and I am really struggled with the above error , appreciate if you can guide me to another troubleshooting

jbrownkramer, 2018-02-18:
@cpq , I agree with @hqawasmi, I can't build on a mac, and mos update latest doesn't work.  I am trying to follow these instructions : https://aws.amazon.com/blogs/apn/aws-iot-on-mongoose-os-part-1/

cpq, 2018-02-18:
Please follow https://forum.mongoose-os.com/discussion/1464/how-to-ask-questions#latest

hqawasmi, 2018-02-19:
its been solved by using Version 1.19.6 , mos update 1.19.6

jbrownkramer, 2018-02-19:
@hqawasmi I did "mos update latest".  Would you expect that to update me to a version of mos that would build the example?

jbrownkramer, 2018-02-19:
@cpq My apologies for not adhering to your standards.  Please understand that I really appreciate the effort that you and other have gone to in order to make this available to the public.  I am just trying to help and to get help in return.

1) My goal is to flash an ESP8266 with the c_mqtt example from this repository using a Mac.

2) I followed the instructions at https://aws.amazon.com/blogs/apn/aws-iot-on-mongoose-os-part-1/ exactly.  In particular, I cloned this repo, went to the directory fw/examples/c_mqtt and issued the command "mos build --arch esp8266".  This failed (I will paste some of the output below).  Following the suggestion on this thread, I tried "mos update latest".  This was yesterday (2/18/2018).  It failed in the same way.

3) I expected the build to not fail.  In particular, I expected the creation of build/fw.zip, which did not happen.

4) Here is the last few lines of the output (which I believe matches the logs):

`/fwbuild-volumes/latest/apps/c_mqtt/esp8266/build_contexts/build_ctx_080118145/src/main.c:85:54: error: left-hand operand of comma expression has no effect [-Werror=unused-value]
LOG(LL_INFO, ("got command: [%.*s]", (int) s->len, s->p));
^
/fwbuild-volumes/latest/apps/c_mqtt/esp8266/build_contexts/build_ctx_080118145/src/main.c:88:5: error: implicit declaration of function 'json_scanf' [-Werror=implicit-function-declaration]
if (json_scanf(s->p, s->len, "{gpio: {pin: %d, state: %d}}", &pin,
^
/fwbuild-volumes/latest/apps/c_mqtt/esp8266/build_contexts/build_ctx_080118145/src/main.c:81:23: error: unused variable 't' [-Werror=unused-variable]
struct json_token t = JSON_INVALID_TOKEN;
^
cc1: all warnings being treated as errors
/mongoose-os/fw/platforms/esp8266/Makefile.build:472: recipe for target '/fwbuild-volumes/latest/apps/c_mqtt/esp8266/build_contexts/build_ctx_080118145/build/objs/main.c.o' failed
make: *** [/fwbuild-volumes/latest/apps/c_mqtt/esp8266/build_contexts/build_ctx_080118145/build/objs/main.c.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes. Stop.
Error: exit status 2
Error: build failed`

hqawasmi, 2018-02-19:
@jbrownkramer, kindly run the command below and test to build again 

mos update 1.19.6 

Please let us know results

cpq, 2018-02-19:
Thanks @jbrownkramer , @hqawasmi 

Please

```
git clone --depth 1 https://github.com/cesanta/mongoose-os
cd mongoose-os/fw/examples/c_mqtt
mos build --platform esp8266
```

Let me know the result.

jbrownkramer, 2018-02-20:
I tried "mos update 1.19.6", and that worked!  Thanks for your help.