cesanta/mongoose-os #375: Can't build for esp32

jptalledo10, 2017-10-29:
Yesterday I open a ticket because I couldn't build for esp32 architecture. I created another VM running Ubuntu 16.04 and I installed mos using apt-get.

The previous fix was to run mos update latest but it seems it is not working:

Running sudo apt update
Hit:1 http://ppa.launchpad.net/mongoose-os/mos/ubuntu xenial InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu xenial InRelease                     
Get:3 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Get:4 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]  
Hit:5 http://security.ubuntu.com/ubuntu xenial-security InRelease              
Fetched 204 kB in 3s (60.8 kB/s)                     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
208 packages can be upgraded. Run 'apt list --upgradable' to see them.
Running sudo apt install --only-upgrade mos
Reading package lists... Done
Building dependency tree       
Reading state information... Done
mos is already the newest version (1.18+f1e3dd6~xenial0).
0 upgraded, 0 newly installed, 0 to remove and 208 not upgraded.

any suggestions to fix it?

I remember somehow some instruction that said to verify the path for ~/.local/xxxx not sure if I am missing a step

keptdream, 2017-10-31:
Same problem:

Connecting to https://mongoose.cloud, user test
[Oct 31 23:55:32.530] Uploading sources (1314 bytes)
Build failed:error response: 400: no such image

dimonomid, 2017-12-13:
This was some transient problem which is resolved now.

closed 2017-12-13 by dimonomid