cesanta/mongoose-os #381: Cannot create custom library

marvinroger, 2017-11-09:
### Context

* Version 1.19.6
* Fresh install
* Reproducible on both MacOS High Sierra and Windows 10

### Issue

From the UI, trying to create or import a library does not do anything. I see a `Repository "mylib" does not exist, cloning...`, but nothing, there's no error.

Trying to create the library manually in the `.mos/libs` directory does not work either. The app cannot find the library (even though it is defined in `libs` as `- name: mylib`).

Creating or importing an app does work.

marvinroger, 2017-11-10:
Actually, I found out that the libraries are indeed cloned, but inside the CWD of the `mos.exe`.  But trying to remote build logs this:

```
Error: reading manifest "/fwbuild-volumes/1.19.6/apps/shutters/esp8266/build_contexts/build_ctx_653396624/libs/mylib/mos.yml": open /fwbuild-volumes/1.19.6/apps/shutters/esp8266/build_contexts/build_ctx_653396624/libs/mylib/mos.yml: no such file or directory
```

Mnark, 2017-11-13:
I have the same issue, but a workaround is to specify the directory using the --libs-dir option when you start mos.exe.
e.g.
C:\> mos.exe --libs-dir <PATH TO MODULES DIRECTORY>.mos\modules-1.19.6

marvinroger, 2017-11-14:
@Mnark indeed, the lib now appears under the "My Libs" section. 
But I still have the `Error: reading manifest "/fwbuild-volumes/1.19.6/apps/shutters/esp8266/build_contexts/build_ctx_653396624/libs/mylib/mos.yml": open /fwbuild-volumes/1.19.6/apps/shutters/esp8266/build_contexts/build_ctx_653396624/libs/mylib/mos.yml: no such file or directory`

Do you build locally or with the remote server?

marvinroger, 2017-11-15:
It does work locally.

dimonomid, 2017-12-12:
Yeah, in order to build project remotely all libs must be accessible (e.g. from github).

closed 2017-12-13 by dimonomid

marvinroger, 2017-12-13:
@dimonomid still, we have to invoke `mos.exe --libs-dir .mos\modules-1.19.6`, otherwise, even locally, it does not work. That's a bug, right?

dimonomid, 2017-12-13:
Ah sorry, actually no, the default location for libs has recently changed to `deps` under the app's dir. So, if you have an app located at `/path/to/myapp`, then `mos build --local` will look for libs in `/path/to/myapp/deps` (and clone them if they don't exist).

marvinroger, 2017-12-13:
Alright, makes sense. But then, what’s the purpose of the “My libs” tab if
there’s no global lib directory?

dimonomid, 2017-12-13:
Fair point, that's a leftover. Will be removed soon.