cesanta/mongoose-os #384: Mongoose OS 1.20- Unable to use mos tool to download file

techparva, 2017-11-17:
With release 1.20, mos tool is updated to get OS and flash separately. It seems there is a bug because of which it's trying to download with version 1.2 instead of 1.20. May be version number is being treated as number internally. 
Following is the log-

[Nov 17 13:03:36.071] Fetching https://github.com/mongoose-os-apps/demo-js/releases/download/1.2/demo-js-esp8266-2M.zip...
Flashing error: failed to load https://github.com/mongoose-os-apps/demo-js/releases/download/1.2/demo-js-esp8266-2M.zip: https://github.com/mongoose-os-apps/demo-js/releases/download/1.2/demo-js-esp8266-2M.zip: failed to fetch: 404 Not Found

bravokeyl, 2017-11-18:
@techparva Try with mos 1.20.1

dimonomid, 2017-12-12:
Yeah it was the case indeed, now it's fixed

closed 2017-12-12 by dimonomid