cesanta/mongoose-os #386: "Save file" on web ui tool makes "fs\init.js" files instead of replacing init.js in Windows

sahilkhanna, 2017-11-23:
I think this is more of a Windows directory path incompatibility issue. The web ui expects a "/" to consider it as another directory.

[image]

Is there any workaround for this?

dimonomid, 2017-12-12:
Sorry, can't reproduce. On the "Device Files" tab, I open `init.js` file, then click "Save File" button, and `init.js` gets saved correctly. Which mos version do you use?

sahilkhanna, 2017-12-13:
I'm sorry, I think I forgot to add some info.
So the issue occurs when you save and copy files from "Projects" tab to the device and NOT on the "Device files" tab. The webui tool confuses with the window's directory naming structure and does not consider "fs\init.js" as a file called "init.js" in "fs" folder but rather treats the whole path "fs\init.js" as a file. This essentially has to do with the backward slash issue in webUI side of things.
FYI I tested this on 
Version: 1.20.1
Build ID: 20171117-113830/1.20.1@d0c3a61b+
OS: Windows 10
Browser: Chrome
but I think the issue still persists in the latest release of mos.exe.
I was able to to edit the .exe file using a hexeditor to essentially replace "\" with "/" which fixed my issue but its not an elegant fix.

dimonomid, 2017-12-13:
Hmm, and still can't reproduce. So I opened `fs\index.html`, clicked the "Save currently opened file" button, and `C:\Users\myuser\.mos\apps-1.20.1\empty-1.15\fs\index.html` file is updated correctly.

[image]

> I was able to to edit the .exe file using a hexeditor to essentially replace "" with "/" which fixed my issue but its not an elegant fix.

Wow that sounds hardcore, however from your explanation `replace "" with "/"` it's unclear what exactly did you replace with what. Also I assume there are plenty of `\` occurrences btw.

sahilkhanna, 2017-12-13:
The issue is not with the save, it's with the "save, copy to device and reboot"  button. Try that and then check your "Device Files" tab.
[image]

dimonomid, 2017-12-13:
Finally it's clear now. :) Ok, I think I'll commit the fix tomorrow, thanks for reporting.

dimonomid, 2017-12-14:
Fixed in private repo, will be published soon.

closed 2017-12-14 by dimonomid