cesanta/mongoose-os #392: Flash of LOLIN32 fails

BluSpanner, 2018-01-04:
I'm unable to flash (via Windows) mongoose-os to a LOLIN32 Lite board
I get the error
`Flashing error: failed to run flasher: failed to talk to flasher: flasher did not respond`
updating mos to 'latest' - still no go...

The board looks OK, as flashing MicroPython to the same board - via esptool = SUCCUSS
mos flash process looks OK, as flashing mongoose-os (via Windows) to different board (the non 'Lite' - LOLIN32 v1.0.0) = SUCCESS

Output from
`mos --logtostderr --esp-baud-rate=115200  flash esp32`
[mos-flash-to-LOLIN32_Lite-failure.txt](https://github.com/cesanta/mongoose-os/files/1602223/mos-flash-to-LOLIN32_Lite-failure.txt)

torfbolt, 2018-01-21:
Same problem here on Debian Linux with mos 1.24:
Flashing to the Lolin32 Lite board fails for all settings and baud rates, usually with the same error @BluSpanner reported.

It works for the identical board with esptool @ 115200 baud. Also all Lolin32 Lite boards I have behave the same.

Spiritdude, 2018-01-26:
Ubuntu 16.04 LTS here also with LOLIN32 Lite, `mos` flashing is unreliable for me, it fails with the same error message as mentioned, or it stalls without giving information how far it flashed, on additional attempts it eventually succeeds for me.

sisao, 2018-03-25:
This worked for me: https://github.com/cesanta/mongoose-os/issues/408#issuecomment-369693237

closed 2018-05-15 by rojer