cesanta/mongoose-os #393: failed to set aws.thing_name

toabhisheksingh, 2018-01-10:
When we run the mos aws-iot-setup command as described in https://aws.amazon.com/blogs/apn/aws-iot-on-mongoose-os-part-1/, we get the following error

Error: failed to set aws.thing_name: no config value at path "aws.thing_name"

tefan57, 2018-01-14:
I had this same issue while trying to build using c, turns out you need to add

- origin: https://github.com/mongoose-os-libs/aws

in libs.

Adding this line removed that exact error.
Hope it helps!

charles-d-burton, 2018-02-10:
This didn't work for me.

tjm1, 2018-02-12:
I am getting the same error.  I see "Adding this line removed that exact error."  what line was added and where was it added? I am confused.

Burt-Silverman, 2018-02-12:
Tim,
Look at the mos.yml file in the top level of the example (c_mqtt directory). You will see origin lines under libs. Add the one that tefan57 metions above.

tjm1, 2018-02-12:
Burt
This did not work for me. I get the same error Like Charles earlier.

Burt-Silverman, 2018-02-12:
Do not forget to reflash after rebuilding. It is easy to make this mistake.

mike-rogers, 2018-02-13:
I ran into this too. To be clear, the solution is to modify your `mos.yaml`. Look for the section that begins:

```yaml
libs:
  - origin: https://github.com/mongoose-os-libs/ca-bundle
  - origin: https://github.com/mongoose-os-libs/http-server
  - origin: https://github.com/mongoose-os-libs/ota-http-server
  - origin: https://github.com/mongoose-os-libs/rpc-loopback
  - origin: https://github.com/mongoose-os-libs/rpc-mqtt
  # other libs...
```

and add the following line:

```yaml
  - origin: https://github.com/mongoose-os-libs/aws
```

Then (per @Burt-Silverman) remember to `mos build` and `mos flash`.

tjm1, 2018-02-13:
mike, Burt. yes you are correct. It works for me when I reflash DUH!. I am a newbie to ESP8266 tools and this was a rookie mistake. Thanks for the help and on to the next challenge.

jcleve, 2018-03-28:
This worked for my ESP32 as well. Thanks everyone!

chrisdowey, 2018-05-12:
Worked for me too, thanks!

PR #506 (2019-05-22): [Fix #393] Failed to set aws.thing_name 
Fix  #393 by adding  `origin: https://github.com/mongoose-os-libs/aws` to avoid this [problem](https://github.com/cesanta/mongoose-os/issues/393)

> failed to set aws.thing_name
commit aee6b1db (2019-05-22): [Fix #393] Failed to set aws.thing_name  (#506)

* Fix failed to set aws.thing_name
closed 2019-05-22 by rojer