cesanta/mongoose-os #394: `mg_sock_addr_to_str()` truncates output in worst case

kzyapkov, 2018-01-11:
If all octets of the device IP contain 3 decimal digits in dot-notation the last digit is missing from the resulting string.

The following is a working fix:

```
diff --git a/mongoose/mongoose.c b/mongoose/mongoose.c
index f9f5d878..f049b4b2 100644
--- a/mongoose/mongoose.c
+++ b/mongoose/mongoose.c
@@ -10127,7 +10127,7 @@ int mg_sock_addr_to_str(const union socket_address *sa, char *buf, size_t len,
       goto cleanup;
     }
 #else
-    if (inet_ntop(AF_INET, (void *) &sa->sin.sin_addr, buf, len - 1) == NULL) {
+    if (inet_ntop(AF_INET, (void *) &sa->sin.sin_addr, buf, len) == NULL) {
       goto cleanup;
     }
 #endif
```

rojer, 2018-01-15:
i think `- 1` here is to ensure NUL-termination. i think we should adjust buffer size at the call site instead. where is the call site for this particular conversion?

kzyapkov, 2018-01-15:
We're using `mgos_net_ip_to_str()` from `mgos_net.c` which sets len=16.

Worst case contains 4 octets times 3 digits each plus 3 dots = 15, so len=16 includes the null byte.

kzyapkov, 2018-02-13:
Hey guys, this is still reproducible

rojer, 2018-02-13:
sorry, this fell through a crack somehow. fix sent.

closed 2018-02-13 by cpq