cesanta/mongoose-os #402: Server Error: {"error":"gcloud device create: : exit status 1"}

mebohac, 2018-02-09:
hi,
i created a new project following this tutorial:  [https://mongoose-os.com/docs/cloud_integrations/gcp.html](url)

and setup the gcp cloud side fine, but upon setting up the device side to register esp32 on cloud iot core, I got this error:  
**> Server Error: {"error":"gcloud device create: : exit status 1"}**

Code:
`[Feb  9 00:29:03.953] Tick uptime: 58.570246 {"free_ram":141412,"total_ram":229632} 

[Feb  9 00:29:04.953] Tock uptime: 59.570258 {"free_ram":141412,"total_ram":229632} 

[Feb  9 00:29:05.935] Connecting to the device...

[Feb  9 00:29:05.953] Tick uptime: 60.569948 {"free_ram":141408,"total_ram":229628} 

[Feb  9 00:29:06.200]   esp32 30AEA43758BC running demo-js

[Feb  9 00:29:06.758] gcp-esp32_3758BC.priv.pem gcp-esp32_3758BC.pub.pem exist, skipping generation

[Feb  9 00:29:06.758] (Re)creating the cloud device...

[Feb  9 00:29:06.953] Tock uptime: 61.570114 {"free_ram":141420,"total_ram":229628} 

[Feb  9 00:29:07.952] Tick uptime: 62.570004 {"free_ram":141412,"total_ram":229632} 

[Feb  9 00:29:08.652] gcloud device deletion: 

[Feb  9 00:29:08.954] Tock uptime: 63.570977 {"free_ram":141412,"total_ram":229632} 

[Feb  9 00:29:09.953] Tick uptime: 64.569948 {"free_ram":141412,"total_ram":229632} 

Server Error: {"error":"gcloud device create: : exit status 1"}

[Feb  9 00:29:10.952] Tock uptime: 65.569994 {"free_ram":141412,"total_ram":229632} 

[Feb  9 00:29:11.953] Tick uptime: 66.569966 {"free_ram":141412,"total_ram":229632} 

`

wondering where the best place is to start to troubleshoot?

thank you!

christianmue1980a, 2018-02-18:
I have exactly the same issue.

lanslans, 2018-02-18:
Same issue here. I've done tons of searching on this, and can't figure it out. Suggestions?

mebohac, 2018-02-19:
yes!  funny you asked.  just finished writing a blog about it today.  hope it helps!

https://myventureintogcp.blogspot.com/

christianmue1980a, 2018-02-19:
Thanks for your post! @mebohac 
I tried your approach, unfortunately the device does not send any data via mqtt to the cloud and vice versa. Can you maybe paste your mqtt configuration here (MOS UI -> Device Config)? Thanks!

WS2133, 2018-03-16:
gcloud components install beta
should fix it

mathias4github, 2018-04-02:
same issue with me, "cloud components install beta" does not help.