cesanta/mongoose-os #403: bcn_timout,ap_probe_send_start occurs very often

honneygoyal, 2018-02-09:
Wifi keep getting reconnected every 3-4 min.
Whenever i try both ap and sta mode this works fine but when only sta mode it keeps getting reconnecting.

ogajvoronsky, 2018-02-09:
I can confirm this behavior... I'm played with power, wifi settings, channels - no luck. It reconnecting every 3-5min

honneygoyal, 2018-02-10:
On esp-32 this works fine but error occurs in esp8266 in station mode.

pontius-g, 2018-06-30:
Same behaviour  with esp8266!
Found a pseudo-fix with persistent ping from another host, it keeps connection stable.
It makes me feel, there is some "unconfigurable power management" problems.

pontius-g, 2018-07-01:
diagnosed with another AP - problem gone.
So, my issue comes with a chinese Wifi "specific beacon timers", guess.

rojer, 2018-08-20:
i believe this is a regression in ESP SDK reported in https://github.com/espressif/ESP8266_NONOS_SDK/issues/119 and as a workaround we rolled back to earlier SDK version in https://github.com/cesanta/mongoose-os/commit/3212e004e4781122a71b4eb267c6f5feac31e686

closed 2018-08-20 by rojer