cesanta/mongoose-os #404: BROKEN: RPC + digest

Andresteve07, 2018-02-16:
Until mid january everything worked just fine. But now, after a few updates on mos, when executing a call using the MG-RPC frame format as described in the docs.
``$ curl --digest "username:password" http://mosIP/rpc/ -d '{"method": "Admin.MyRPC", "args":{} , "tag": "MY_TAG", "id":12345 }'``
Returns 
``Invalid request``
In the serial log:
``mg_rpc_ev_handler    0x3fff1074 INVALID FRAME (0): ``

But if we run the call with the RPC name at the end of the path it works flawlessly.
``$ curl --digest "username:password" http://mosIP/rpc/Admin.MyRPC``
Sadly enough throughout our entire client application we use the former request format. Is there any hope you can attend to this issue?

The acl file content would be:
``[
    {
        "method": "Admin.*,OTA.*",
        "acl": "username"
    },
    {
        "method": "*",
        "acl": "-*"
    }
]``
The auth file content would be:
``username:myapp:caa845200e3618d5592f9986eadc9068``

cpq, 2018-02-16:
Thanks @Andresteve07 ! Fortunately, all sources are open and you can find out why you're getting  `Invalid request`. From our point of view, everything works as intended.

Andresteve07, 2018-02-16:
Thank you for the rapid response. We'll examine the code then.

closed 2018-02-16 by Andresteve07