cesanta/mongoose-os #408: ESP32-WROOM (Geekform) unable to flash

Morlack, 2018-03-01:
Hey all,

I am currently experiencing issues with mongoose os. It works fine for my esp8266 boards, but it does not work for esp32.

Flashing it using the arduino IDE does work, and I have no clue why it doesn't with mongooseOS. 

This is the output of `mos flash --port /dev/tty.wchusbserial14120 --logtostderr -v 4`

```
Loaded demo-js/esp32 version 1.0 (20180301-083741/???)
I0301 10:09:30.039876    2692 ourutil.go:19] Loaded demo-js/esp32 version 1.0 (20180301-083741/???)
I0301 10:09:30.039975    2692 flash_params.go:104]  -> -1, -1, -1
Opening /dev/tty.wchusbserial14120 @ 115200...
I0301 10:09:30.039993    2692 report.go:13] Opening /dev/tty.wchusbserial14120 @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
I0301 10:09:30.067504    2692 report.go:13] Connecting to ESP32 ROM, attempt 1 of 10...
I0301 10:09:31.696938    2692 rom_client.go:180] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0301 10:09:31.697036    2692 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0301 10:09:31.903692    2692 rom_client.go:180] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0301 10:09:31.903755    2692 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0301 10:09:31.911192    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.911290    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.911358    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.911373    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.913991    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.914030    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.914117    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.914131    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.916885    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.916926    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.917020    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.917038    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.940080    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.940120    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.940220    2692 slip.go:46] <= (12) 010804000712205500000000
I0301 10:09:31.940234    2692 rom_client.go:224] <= {cmd:Sync(8) value:1428165127 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.952068    2692 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"04a10160"}
I0301 10:09:31.952124    2692 slip.go:90] => (12) 000a04000000000004a10160
I0301 10:09:31.977054    2692 slip.go:46] <= (12) 010a04000000000000000000
I0301 10:09:31.977087    2692 rom_client.go:224] <= {cmd:ReadReg(10) value:0 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:31.977118    2692 rom_client.go:435] ReadReg(0x6001a104) => 0x00000000
I0301 10:09:31.977132    2692 efuses.go:407] 00000000
I0301 10:09:31.990767    2692 rom_client.go:180] => {cmd:WriteReg(9) csum:0 arg(16):"fca00160a55a0000ffffffff00000000"}
I0301 10:09:31.990870    2692 slip.go:90] => (24) 0009100000000000fca00160a55a0000ffffffff00000000
I0301 10:09:32.016827    2692 slip.go:46] <= (12) 010904000000000000000000
I0301 10:09:32.016872    2692 rom_client.go:224] <= {cmd:WriteReg(9) value:0 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:32.016916    2692 rom_client.go:449] WriteReg(0x6001a0fc, 0x00005aa5)
I0301 10:09:32.030471    2692 rom_client.go:180] => {cmd:WriteReg(9) csum:0 arg(16):"04a1016001000000ffffffff00000000"}
I0301 10:09:32.030528    2692 slip.go:90] => (24) 000910000000000004a1016001000000ffffffff00000000
I0301 10:09:32.056592    2692 slip.go:46] <= (12) 010904000000000000000000
I0301 10:09:32.056654    2692 rom_client.go:224] <= {cmd:WriteReg(9) value:0 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:32.056684    2692 rom_client.go:449] WriteReg(0x6001a104, 0x00000001)
I0301 10:09:32.068296    2692 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"04a10160"}
I0301 10:09:32.068335    2692 slip.go:90] => (12) 000a04000000000004a10160
I0301 10:09:32.093440    2692 slip.go:46] <= (12) 010a04000000000000000000
I0301 10:09:32.093483    2692 rom_client.go:224] <= {cmd:ReadReg(10) value:0 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:32.093509    2692 rom_client.go:435] ReadReg(0x6001a104) => 0x00000000
I0301 10:09:32.093524    2692 efuses.go:407] 00000000
I0301 10:09:32.093535    2692 efuses.go:435] eFuseCtlOpRead successful
I0301 10:09:32.093590    2692 efuses.go:117] Reading eFuse block 0 @ 0x6001a000...
I0301 10:09:32.107546    2692 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"00a00160"}
I0301 10:09:32.107641    2692 slip.go:90] => (12) 000a04000000000000a00160
I0301 10:09:32.132676    2692 slip.go:46] <= (12) 010a04000000000000000000
I0301 10:09:32.132710    2692 rom_client.go:224] <= {cmd:ReadReg(10) value:0 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:32.132738    2692 rom_client.go:435] ReadReg(0x6001a000) => 0x00000000
I0301 10:09:32.144452    2692 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"04a00160"}
I0301 10:09:32.144497    2692 slip.go:90] => (12) 000a04000000000004a00160
I0301 10:09:32.169365    2692 slip.go:46] <= (12) 010a0400981521a400000000
I0301 10:09:32.169398    2692 rom_client.go:224] <= {cmd:ReadReg(10) value:2753631640 ok:true lastError:0 body(4):"00000000"}
I0301 10:09:32.169420    2692 rom_client.go:435] ReadReg(0x6001a004) => 0xa4211598
[... 207 lines cut ...]
I0301 10:09:34.217560    2692 flasher_client.go:223] <= c0fa96cdc0c0
I0301 10:09:34.318813    2692 slip.go:90] => (0) 
I0301 10:09:34.318905    2692 flasher_client.go:103] Echo(8) [1583383712]
I0301 10:09:34.319009    2692 slip.go:90] => (1) 08
I0301 10:09:34.319122    2692 slip.go:90] => (4) a084605e
I0301 10:09:34.341948    2692 flasher_client.go:223] <= c023a005c000c0
I0301 10:09:34.443195    2692 slip.go:90] => (0) 
I0301 10:09:34.443252    2692 flasher_client.go:103] Echo(8) [3005454171]
I0301 10:09:34.443284    2692 slip.go:90] => (1) 08
I0301 10:09:34.443327    2692 slip.go:90] => (4) 5b9723b3
I0301 10:09:34.466161    2692 flasher_client.go:223] <= c0dbd99a03c080
I0301 10:09:34.567426    2692 slip.go:90] => (0) 
I0301 10:09:34.567557    2692 flasher_client.go:103] Echo(8) [1614498188]
I0301 10:09:34.567587    2692 slip.go:90] => (1) 08
I0301 10:09:34.567628    2692 slip.go:90] => (4) 8c493b60
I0301 10:09:34.590340    2692 flasher_client.go:223] <= c0ac6d60c000c0
I0301 10:09:34.693195    2692 main.go:241] Error: /Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:233: flasher did not respond
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:95: failed to talk to flasher
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:67: 
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/common.go:65: failed to run flasher
/Users/mos/go/src/cesanta.com/mos/flash/esp/flasher/flash.go:43: 
/Users/mos/go/src/cesanta.com/mos/flash.go:141: 
/Users/mos/go/src/cesanta.com/mos/main.go:137: 
Error: failed to run flasher: failed to talk to flasher: flasher did not respond
```

Morlack, 2018-03-01:
Also not I have tried every baud rate, pulling GPIO0 and esp-flash params with QIO instead of DIO.

cpq, 2018-03-01:
Did you try `--esp-baud-rate=0` ?

Morlack, 2018-03-25:
Sorry for the late reply, but it worked! 
Just my curiosity, why does this work, and other baud rates do not?

closed 2018-05-15 by rojer