cesanta/mongoose-os #413: JS Console in IDE to run simple snippets and commands

chall3ng3r, 2018-03-13:
Been tinkering with this awesome framework for a bit, and today finally had a chance to do a quick PoC using it for a client.

While there is simple and good API reference, and examples, which are good to just copy or look for API usage examples. But sometimes, it helps if developer can run small part of the code to see the result, before putting it in a method.

It'd be awesome if MongooseOS add an option for "JS Console" window beside "Output" window, this can allow to run a small bit of JS code on connected device. I believe this will improve ease of testing APIs quickly before adding in complicated code.

cpq, 2018-05-15:
That sounds good!

You can write a library that adds an RPC service `JS.Exec`.
That lib creates an mJS instance, executes a passed string, and kills the instance (optionally calling cleanup function if defined).

If you write such a library and contribute to the community, we'll integrate it into the UI (or you can do it yourself, since mos-tool is open source too).
Are you up for it?

tripflex, 2018-06-08:
@cpq you mentioned this in the gitter chat, but can't seem to get it to work with params .. can you provide an example for this please?

```js
RPC.addHandler('JS.exec', function(args) {
  let i = 0, j, name = args.func || '', obj = global;
  while (true) {
    j = name.indexOf('.', i);
    if (j < 0) break;
    print(JSON.stringify({a: name.slice(i, j)}));
    obj = obj[name.slice(i, j)];
    i = j + 1;
  }
  obj = obj[name.slice(i)];
  obj.apply(args.params);
});
```

chall3ng3r, 2018-06-09:
@cpq sorry about late reply. Yes can do it, been busy with client projects recently. Will be free for a bit soon and will try it out and see how it goes.

cpq, 2018-06-09:
Yeah, that `JS.exec` is not a "full" REPL, cause it cannot create new pieces of code. It can only execute existing functions.