cesanta/mongoose-os #420: MAC address of connected device

ISTechSolutions, 2018-04-20:
When a device connects to the ESP32 via wifi, the following is displayed:

I[Apr 20 11:05:10.983]  (807722) wifi: n:6 0, o:6 0, ap:6 1, sta:255 255, prof:6
I (807732) wifi: station: 2c:0e:3d:c1:7e:3a join, AID=1, g, 20
e[Apr 20 11:05:11.000] sp32_wifi_ev        WiFi AP: station 2C0E3DC17E3A (aid 1) connected
mongoose_poll        New heap free LWM: 177260

It seems that there is no method to obtain the MAC address afterwards. Is it possible to add an event similar to:
	mgos_event_add_group_handler(MGOS_EVENT_GRP_NET, net_cb, NULL);

It should both report when the connection is established and disconnected. Very useful to identify a user or to only allow registered users.

kzyapkov, 2018-04-20:
I think this is about the MAC address of the client, and this is at least also valid on esp8266.

ISTechSolutions, 2018-04-22:
I need the MAC address of the client (mobile phone in my case) not the server (ESP32).

rojer, 2018-08-20:
fixed by https://github.com/mongoose-os-libs/wifi/commit/35277c2cbd3f7e638685018b154873203cdc0bd0

closed 2018-08-20 by rojer