cesanta/mongoose-os #423: Built-in LED does not light up on LOLIN32

stefan-girlich, 2018-04-29:
I recently used Mongoose OS to flash the _demo-js_ example firmware onto a new LOLIN32 by Wemos. By factory default, the built-in LED used to blink, but it stopped blinking as soon as installed the firmware via the Mongoose web interface and I was not able to turn it on again since.

Apart from a sudden hardware defect I can only imagine that the LOLIN32 would require specific Mongoose OS support which is not available yet according to the [supported hardware page](https://mongoose-os.com/docs/reference/supported-hardware.html).
So far I tried changing the value of _led_ to _5_ in the JavaScript code (see [here](https://github.com/espressif/arduino-esp32/issues/359), but without visible success: The [scheduled function](https://github.com/mongoose-os-apps/demo-js/blob/master/fs/init.js#L25) is executed and the debug information is printed, but the LED does never turn on.

Any ideas? I am considering trying out Micropython to see if the board itself is intact, but I'm looking forward to further recommendations.

### Stats
MacOS 10.11.6
Chrome 65
Wemos LOLIN32 V1.0.0 ESP32 board

nliviu, 2018-04-30:
After changing the line `let led = Cfg.get('pins.led');` to `let led = 5;//Cfg.get('pins.led');` and rebooting the board, the led blinks as expected.

I don't have any problems using Mongoose OS with Lolin32, Lolin32-pro and MH-ET Esp32 Minikit.

stefan-girlich, 2018-04-30:
It now works with the freshly imported and modified _demo-js_ code sample - thanks for clarifying! I think that previously I may have accidentally removed  [line of code](https://github.com/mongoose-os-apps/demo-js/blob/master/fs/init.js#L23) which defines the pin as output.

closed 2018-04-30 by stefan-girlich