cesanta/mongoose-os #424: Missing MGOS_GPIO_INT_EDGE_ANY in the if condition

nkapser, 2018-05-07:
https://github.com/cesanta/mongoose-os/blob/bac8bbfa561f71072b5285fcc0745df5ceab8c5b/fw/src/mgos_gpio.c#L139

When I updated my mos version 2.0+, I noticed that my esp32 board is not responding to INT_EDGE_ANY. I found that you are not checking for the INT_EDGE_ANY and returning false!

Please fix this.

- kapser

cpq, 2018-05-08:
@rojer please fix

rojer, 2018-05-09:
@nkapser that's right. this is due to the change in how debouncing logic works (https://github.com/cesanta/mongoose-os/commit/cb8e3dfb9c1750bd7bd5282e652615032395968e#diff-f725f868172e28918edf54206c820338) - we need to know which is the active state of the button, 0 or 1, so we can check if the button is active after designated debounce time. we deduce it from the edge that is used to detect the activation.
with any edge we just don't know what to check for. so, if you need interrupts on any edge, use `mgos_gpio_set_int_handler()`.

closed 2018-05-09 by rojer

close2, 2018-12-09:
Please add a static assert.