cesanta/mongoose-os #426: OTA rpc reverts

AlessiaPileri, 2018-05-10:
I'm using mongoose-os on ESP32 platform and I have an issue trying to perform OTA update via RPC http (I'm using licensed version of the lib).
After the rpc command, OTA update seems to be ok,as I can see in the mos console:
``updater_finish       Finished: 1 Update applied, finalizing ``
But after few minutes I see this message in mos console:
``OTA failure: reverting``
and firmware version is reverted.

Full log downloaded from ``mos console``:  [log_mos.txt](https://github.com/cesanta/mongoose-os/files/1990934/log_mos.txt)

cpq, 2018-05-11:
Did you `OTA.Commit` the new firmware after boot ?

AlessiaPileri, 2018-05-11:
Should I send OTA.Commit command explicitly? In past versions of the libs, send OTA.Update was enough to update the firmware.

cpq, 2018-05-11:
Please read https://mongoose-os.com/docs/book/ota.html and https://github.com/mongoose-os-libs/rpc-service-ota#otaupdate

cpq, 2018-05-15:
Closing this - feel free to reopen if required.

closed 2018-05-15 by cpq