cesanta/mongoose-os #428: Core Dump Analysis on WINDOWS doesn't work

malotian, 2018-05-16:
**Core Dump Analysis on WINDOWS doesn't work** probably marked bold needs to fixed as well.

[May 16 11:52:04.549] I (3649) wifi: pm start, type: 0
[May 16 11:52:04.549]
[May 16 11:52:04.561] mgos_net_on_change_c WiFi STA: connected
[May 16 11:52:04.578] Event.addGroupHandler.WiFi: {"ip":false,"connected":true}
[May 16 11:52:04.695] I (3789) event: sta ip: 192.168.43.164, mask: 255.255.255.0, gw: 192.168.43.1
[May 16 11:52:04.710] mgos_net_on_change_c WiFi STA: ready, IP 192.168.43.164, GW 192.168.43.1, DNS 192.168.43.1
[May 16 11:52:04.722] Event.addGroupHandler.WiFi: {"ip":true,"connected":true}
[May 16 11:52:05.772] mgos_sntp_query      SNTP query to time.google.com
[May 16 11:52:06.520] mgos_sntp_ev         SNTP reply from 216.239.35.12: time 1526451727.772768, local 8.862162, delta 1526451718.910606
[May 16 11:52:10.478] Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
[May 16 11:52:10.479] . Exception was unhandled.
[May 16 11:52:10.487] Core 0 register dump:
[May 16 11:52:10.489] PC      : 0x400ec3fb  PS      : 0x00060430  A0      : 0x800ea26f  A1      : 0x3ffc2280
[May 16 11:52:10.494] A2      : 0x3ffc23e0  A3      : 0x00000000  A4      : 0x00000008  A5      : 0x3ffc2350
[May 16 11:52:10.504] A6      : 0x3ffd4880  A7      : 0x40180000  A8      : 0x800ec3f9  A9      : 0x3ffc2260
[May 16 11:52:10.513] A10     : 0x3ffc2358  A11     : 0x3ffc2350  A12     : 0x3ffc228c  A13     : 0x3ffc2280
[May 16 11:52:10.520] A14     : 0x3ffc22c8  A15     : 0x3f40e9c8  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
[May 16 11:52:10.529] EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[May 16 11:52:10.530]
[May 16 11:52:10.546] Backtrace: 0x400ec3fb 0x400ea26c 0x400f1262 0x400f56ea 0x400f7a56 0x400f8879 0x400f8a7d 0x400f8c5b 0x400e3849 0x40191cae 0x401930fe 0x40195679 0x4018f482 0x400e2e0f 0x40082e9e 0x40082d8e
[May 16 11:52:10.548] --- BEGIN CORE DUMP ---
[May 16 11:52:10.548] mos: catching core dump
[May 16 11:53:05.085] ---- END CORE DUMP ----
[May 16 11:53:05.138] mos: wrote to C:\Users\harjinder_dhanjal\AppData\Local\Temp\core--ESP32-20180516-115305.939998991 (628279 bytes)
[May 16 11:53:05.163] mos: analyzing core dump
Using ELF file at: D:\malotian\iot\strata.sphere\build\objs\fw.elf
Using Docker image: docker.cesanta.com/esp32-build:3.0-rc1-r8
Using Mongoose OS souces at: D:\malotian\iot\strata.sphere\deps\mongoose-os
Running docker run --rm -v D:\malotian\iot\strata.sphere\build\objs\fw.elf:/fw.elf -v C:\Users\harjinder_dhanjal\AppData\Local\Temp\core--ESP32-20180516-115305.939998991:/core -v D:\malotian\iot\strata.sphere\deps\mongoose-os:/mongoose-os -v D:\malotian\iot\strata.sphere:D:/malotian/iot/strata.sphere docker.cesanta.com/esp32-build:3.0-**rc1-r8** bash -c /usr/local/bin/serve_core.py --rom=/opt/Espressif/rom/rom.bin --rom_addr=0x40000000 --xtensa_addr_fixup=true /fw.elf /core & **$MGOS_TARGET_GDB** /fw.elf -ex 'target remote 127.0.0.1:1234' -ex 'set confirm off' -ex bt -ex quit
docker: Error response from daemon: Mount denied:
The source path "D:/malotian/iot/strata.sphere:D"
doesn't exist and is not known to Docker.
See 'docker run --help'.

cpq, 2018-05-17:
We mostly use Mac and Linux.

Since you're a Windows user, feel free to fix this issue on Windows and contribute that to everybody else.

malotian, 2018-05-17:
hmm, sure thanks will try to, with my limited knowledge.

malotian, 2018-05-17:
> It looks like a docker issue. Is D: shared with docker as /d?

Yes is shared..

malotian, 2018-06-11:
Thanks for updates, switched to linux

closed 2018-06-13 by malotian