cesanta/mongoose-os #431: API docs on website are broken/missing

ricochet1k, 2018-05-31:
Every time I try to click on an API link on [this page](https://mongoose-os.com/libs.html) I get redirected to your homepage, https://mongoose-os.com. This happens even if I type the URL in directly, leading me to think that your site's 404 page is redirecting to the homepage... that's not a good idea. 404 pages are better for people and web crawlers alike.

cpq, 2018-05-31:
Can't actually reproduce.
```
$ curl -s -i https://mongoose-os.com/libs.html | head -10
HTTP/1.1 200 OK
Server: nginx/1.12.2
Date: Thu, 31 May 2018 11:12:47 GMT
Transfer-Encoding: chunked
Connection: keep-alive
X-Powered-By: Express

<!DOCTYPE html>
<html lang="en">
  <head>
```

closed 2018-05-31 by cpq

ricochet1k, 2018-05-31:
Please read more carefully. I said clicking on an API link *on that page*. That page works fine, but the links are broken. See below for the redirect.

```
$ curl -s -i https://mongoose-os.com/api/api_adc.js.html | head -15
HTTP/1.1 302 Found
Server: nginx/1.12.2
Date: Thu, 31 May 2018 14:47:06 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 23
Connection: keep-alive
X-Powered-By: Express
Location: /
Vary: Accept

Found. Redirecting to /
```