cesanta/mongoose-os #432: API docs on website are broken/missing

ricochet1k, 2018-06-08:
Every time I try to click on **an API link** on https://mongoose-os.com/libs.html, such as https://mongoose-os.com/api/api_adc.js.html, I get redirected to your homepage, https://mongoose-os.com. This happens even if I type the URL in directly, leading me to think that your site's 404 page is redirecting to the homepage... that's not a good idea. 404 pages are better for people and web crawlers alike.

```
$ curl -s -i https://mongoose-os.com/api/api_adc.js.html | head -15
HTTP/1.1 302 Found
Server: nginx/1.12.2
Date: Thu, 31 May 2018 14:47:06 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 23
Connection: keep-alive
X-Powered-By: Express
Location: /
Vary: Accept

Found. Redirecting to /
```

After doing some more digging, it looks like `/libs.html` on your site seems to be abandoned for the `/docs/` directory. But `/libs.html` is result 2 when doing a google search for `mongoose os api`, and having pages that high up in google results that *don't work* is going to drive users away.

You used to have these nice looking API docs in the /docs/ directory, but I guess you're going away from them? https://mongoose-os.com/docs/api/api_config.js.html now returns `Error: ENOENT: no such file or directory, stat '/data/mongoose-os-docs/api/api_config.js.html'`, shouldn't this also be a 404 page and not an error?

Also, yes, this is a duplicate of #431 because the original issue was closed after a hasty attempt at reproducing the issue based on a misunderstanding of what actually happened. May I suggest that you don't close bugs so quickly without giving people a chance to clarify?

iomik, 2018-06-09:
Same issue here. I need to check onewire api docs

paulolc, 2018-09-02:
It seems the api documentation is being done  inline in the api source code comments. Maybe it's a missing apidoc generation step somewhere in https://github.com/cesanta/mongoose-os-docs? Should this issue be raised in mongoose-os-docs instead?

cpq, 2018-09-14:
Finished refacroring API docs.
Some individual bits of the reference require work, but the overall structure is complete.
https://mongoose-os.com/docs/api/core/adc.md

closed 2018-09-14 by cpq