cesanta/mongoose-os #433: Interrrupts are skipped when occurring with high processing load

mamuesp, 2018-06-09:
@cpq - As discussed in Gitter, it would be helpful to increase the amount of the INT queue so that there might be a balance between INT "flooding" and INT "ignoring" when the system is busy otherwise. You asked for an issue to may set the right prio, so here it is! ;-)

mamuesp, 2019-01-03:
Yeah, it's good to bring that in mind, because interrupts and MJS are not really friends these days ... ;-)

cpq, 2019-01-03:
PR please!

mamuesp, 2019-01-03:
@cpq - We discussed that (I believe @rojer was into the discussion as well, if I remember correctly), and from your side came the question to post an issue so that you or @rojer will look after this problem as soon as possible (this was the promise then ...) ;-) - because it's not an excpected behaviour that interruts are going down the drain in MJS. So this is no complaining about missing features but a really (nearly) faulty behaviour and a fix was proposed from your (or @rojer's) side. So how could I provide a PR at this place? There was also a concrete idea to pass a parameter when settings the interrupt handler so one may increase the queue for the interrupts.

rojer, 2019-01-11:
i believe i have a fix for this, about to send a PR

closed 2019-01-11 by rojer