cesanta/mongoose-os #435: Flash encryption not working

dude-mansaa, 2018-07-03:
#mos flash esp32
This command is flashing the demo.js in the esp32.

#mos flash-read **--arch** esp32 0x190000 2000 - //command changed
#mos flash-read --platform esp32 0x190000 2000 -

This command is reading 2000 bytes of demo.js which the CA certificate.
C:\Users\Mansaa\.mos\apps-2.3\wifi_bulb>mos flash-read --platform esp32 0x190000 2000 -
Using port COM9
Opening COM9 @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
  Connected, chip: ESP32D0WDQ6 R1
Running flasher @ 0...
  Flasher is running
Reading 2000 @ 0x190000...
Read 2000 bytes in 0.20 seconds (76.97 KBit/sec)
�              ������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������)  �  �  �I  ca.pem                                          ���������     
                  ! " # $ % & ' ( ) * + , - . / 1 2 3 4 5 6 7 8 9 : ; < = > ? A B C D E F G H I J K L M N O Q R ����������������������������������   �Subject: O=Digital Signature Trust Co., CN=DST Root CA X3
Not Before: Sep 30 21:12:19 2000 GMT
Not After : Sep 30 14:01:15 2021 GMT
-----BEGIN CERTIFICATE-----
MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/
MSQwIgYDVQQKExtEaWdpdGFsIF  �NpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow
PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD
Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB
AN+v6ZdQCINXtMxiZ  �faQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O
rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq
OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b
xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw
7BZy1Sbs  �OFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD
aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV
HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG
SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69  �
ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr
AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz
R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5
JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06Xy  �xV3bqxbYo
Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ
-----END CERTIFICATE-----

Subject: C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
Not Before: May 12 18:46:00 2000 GMT
Not After : May

#mos -X esp32-gen-key flash_encryption_key fe.key \ --esp32-enable-flash-encryption --dry-run=false
This command is generating the fe.key and enabling flash encryption.
C:\Users\Mansaa\.mos\apps-2.3\wifi_bulb>mos -X esp32-gen-key flash_encryption_key fe.key \
Using port COM9
Opening COM9 @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
Connecting to ESP32 ROM, attempt 2 of 10...
Connecting to ESP32 ROM, attempt 3 of 10...
Connecting to ESP32 ROM, attempt 4 of 10...
Connecting to ESP32 ROM, attempt 5 of 10...
  Connected, chip: ESP32D0WDQ6 R1
Device MAC address: b4:e6:2d:93:a6:25
Error: flash_encryption_key is already set

#mos flash esp32 --esp32-encryption-key-file fe.key
This command is flashing the demo.js but with encryption key
C:\Users\Mansaa\.mos\apps-2.3\wifi_bulb>mos flash esp32 --esp32-encryption-key-file fe.key
Fetching https://github.com/mongoose-os-apps/demo-js/releases/download/2.3/demo-js-esp32.zip...
Loaded demo-js/esp32 version 1.0 (20180615-171315/???)
Using port COM9
Opening COM9 @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
  Connected, chip: ESP32D0WDQ6 R1
Running flasher @ 0...
  Flasher is running
Flash size: 4194304, params: 0x0220 (dio,32m,40m)
Deduping...
    16384 @ 0x9000 -> 8192
   262144 @ 0x190000 -> 0
Writing...
    20480 @ 0x1000
     4096 @ 0x8000
     8192 @ 0x9000
     8192 @ 0xd000
  1540096 @ 0x10000
Wrote 1573952 bytes in 137.11 seconds (89.68 KBit/sec)
Verifying...
    18368 @ 0x1000
     3072 @ 0x8000
    16384 @ 0x9000
     8192 @ 0xd000
  1536128 @ 0x10000
   262144 @ 0x190000
Booting firmware...
All done!

But now when again I am performing 
#mos flash-read --arch esp32 0x190000 2000 -
It is showing

C:\Users\Mansaa\.mos\apps-2.3\wifi_bulb>mos flash-read --platform esp32 0x190000 2000 -
Using port COM9
Opening COM9 @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
  Connected, chip: ESP32D0WDQ6 R1
Running flasher @ 0...
  Flasher is running
Reading 2000 @ 0x190000...
Read 2000 bytes in 0.20 seconds (76.97 KBit/sec)
�              ������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������)  �  �  �I  ca.pem                                          ���������     
                  ! " # $ % & ' ( ) * + , - . / 1 2 3 4 5 6 7 8 9 : ; < = > ? A B C D E F G H I J K L M N O Q R ����������������������������������   �Subject: O=Digital Signature Trust Co., CN=DST Root CA X3
Not Before: Sep 30 21:12:19 2000 GMT
Not After : Sep 30 14:01:15 2021 GMT
-----BEGIN CERTIFICATE-----
MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/
MSQwIgYDVQQKExtEaWdpdGFsIF  �NpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow
PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD
Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB
AN+v6ZdQCINXtMxiZ  �faQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O
rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq
OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b
xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw
7BZy1Sbs  �OFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD
aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV
HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG
SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69  �
ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr
AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz
R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5
JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06Xy  �xV3bqxbYo
Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ
-----END CERTIFICATE-----

Subject: C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
Not Before: May 12 18:46:00 2000 GMT
Not After : May

and in mos tool I am getting error
[Jul  3 17:46:15.821] ets_main.c 371 

[Jul  3 17:46:16.156] ets Jun  8 2016 00:22:57

[Jul  3 17:46:16.158] 

[Jul  3 17:46:16.158] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

[Jul  3 17:46:16.163] flash read err, 1000

[Jul  3 17:46:16.165] ets_main.c 371 

[Jul  3 17:46:16.500] ets Jun  8 2016 00:22:57

[Jul  3 17:46:16.503] 

[Jul  3 17:46:16.503] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

[Jul  3 17:46:16.508] flash read err, 1000

[Jul  3 17:46:16.510] ets_main.c 371 

[Jul  3 17:46:16.845] ets Jun  8 2016 00:22:57

[Jul  3 17:46:16.847] 

[Jul  3 17:46:16.848] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

[Jul  3 17:46:16.853] flash read err, 1000

[Jul  3 17:46:16.855] ets_main.c 371 

[Jul  3 17:46:17.190] ets Jun  8 2016 00:22:57

[Jul  3 17:46:17.192] 

[Jul  3 17:46:17.192] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

[Jul  3 17:46:17.197] flash read err, 1000

Please provide the solution regarding how to do the flash encryption and if there is an error in the commands.

rojer, 2018-08-20:
this is because filesystem is handled slightly differently: first flashing is in plain text but it is then encrypted on first boot. it looks like you did not allow that process to take place.

closed 2018-08-20 by rojer

dude-mansaa, 2018-08-20:
I have followed the process of encryption. I do not understand what you are saying. Could you please explain?

dude-mansaa, 2018-08-20:
https://forum.mongoose-os.com/discussion/2627/esp32-flash-encryption-failure

You can even check this link for better logs.

dude-mansaa, 2018-08-20:
https://forum.mongoose-os.com/discussion/comment/7025/#Comment_7025

You can even check this link for better logs.